arendst/Tasmota #5167: NodeMCU with WS2812

nromensky, 2019-02-08:
QUESTION

i searched issues and googled, but can't find definitive answer to my problem. WS2812 works only on GPIO3{RX} with USE_WS2812_DMA, any other pin does not work, and it does not work without USE_WS2812_DMA, and i would be happy to use it like this, but when i enable scheme it freezes, LED plays the scheme, but i loose access. I use 30 pixel LED with external 2A PSU, grounds are connected.

I use 6.4.1.14, in 6.4.1 it does not work at all, even with DMA

please help

Jason2866, 2019-02-08:
Try dev builds with Core 2.5.0 from http://thehackbox.org/tasmota/020500/sonoff.bin

ascillato2, 2019-02-08:
Hi,

Sorry, I can't reproduce your issue. I'm actually using a standard WS2812 on different gpios without DMA and works fine. I'm using a strip of 100 leds and 1 power supply of 2A 5V

Can you erase all flash with esptool.py as explained in the wiki and flash again? 
Also, can you complete the troubleshooting template in order to have more information so as to properly help you? 
Thanks

nromensky, 2019-02-08:
> http://thehackbox.org/tasmota/020500/sonoff.bin

thx Jason! I've tried http://thehackbox.org/tasmota/020500/sonoff-4m.bin , same result
actually I use Arduino IDE with 2.5.0

nromensky, 2019-02-08:
> Can you erase all flash with esptool.py as explained in the wiki and flash again?

I've done erase

```
C:\Windows\system32>esptool.py --port COM4 erase_flash
esptool.py v2.6
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 80:7d:3a:7f:db:e6
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.0s
Hard resetting via RTS pin...
```

> Also, can you complete the troubleshooting template in order to have more information so as to properly help you?

here's template

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE with esp8266 v2.5.0
- [x] Hardware used : NodeMCU v3 with 4M flash
- [-] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : 6.4.1.14
- [x] Provide the output of command ``status 0`` :

```
16:26:36 CMD: status 0
16:26:36 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["sonoff-NodeMCU"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:26:36 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:15","StartupUTC":"2019-02-08T15:26:21","Sleep":50,"BootCount":7,"SaveCount":17,"SaveAddress":"3FB000"}}
16:26:36 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.14(sonoff)","BuildDateTime":"2019-02-08T15:14:24","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
16:26:36 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Charles_Darwin",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
16:26:36 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":3476,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","0FFAE794","000383A0","23B617CE","00003BC0"]}}
16:26:36 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7142","IPAddress":"192.168.1.63","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"80:7D:3A:7F:DB:E6","Webserver":2,"WifiConfig":4}}
16:26:36 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7FDBE6E6","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:26:36 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 08 15:26:36 2019","Local":"Fri Feb 08 16:26:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:10","Sunset":"17:58"}}
16:26:36 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-08T16:26:36"}}
16:26:36 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-08T16:26:36","Uptime":"0T00:00:15","Vcc":2.998,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":10,"Color":"191919","HSBColor":"0,0,10","Channel":[9,9,9],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Charles_Darwin","BSSId":"64:F6:9D:D5:E7:71","Channel":1,"RSSI":74}}}
```

nromensky, 2019-02-08:
here's status 0 for firmware complied with USE_WS2812_DMA, WS2812 connected to RX and works fine, except Schemes {it freezes once i start them}

```
16:58:47 CMD: status 0
16:58:47 MQT: stat/sonoff-NodeMCU/STATUS = {"Status":{"Module":18,"FriendlyName":["sonoff-NodeMCU"],"Topic":"sonoff-NodeMCU","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:58:47 MQT: stat/sonoff-NodeMCU/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:30","StartupUTC":"2019-02-08T15:47:17","Sleep":50,"BootCount":14,"SaveCount":50,"SaveAddress":"3FA000"}}
16:58:47 MQT: stat/sonoff-NodeMCU/STATUS2 = {"StatusFWR":{"Version":"6.4.1.14(sonoff)","BuildDateTime":"2019-02-08T15:37:35","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
16:58:47 MQT: stat/sonoff-NodeMCU/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Charles_Darwin",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
16:58:47 MQT: stat/sonoff-NodeMCU/STATUS4 = {"StatusMEM":{"ProgramSize":597,"Free":3476,"Heap":11,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","0FFBE794","000383A0","23B617CE","00003BC0"]}}
16:58:47 MQT: stat/sonoff-NodeMCU/STATUS5 = {"StatusNET":{"Hostname":"sonoff-NodeMCU-7142","IPAddress":"192.168.1.63","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"80:7D:3A:7F:DB:E6","Webserver":2,"WifiConfig":4}}
16:58:47 MQT: stat/sonoff-NodeMCU/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.243","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7FDBE6E6","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:58:49 MQT: stat/sonoff-NodeMCU/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 08 15:58:47 2019","Local":"Fri Feb 08 16:58:47 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:10","Sunset":"17:58"}}
16:58:49 MQT: stat/sonoff-NodeMCU/STATUS10 = {"StatusSNS":{"Time":"2019-02-08T16:58:49"}}
16:58:49 MQT: stat/sonoff-NodeMCU/STATUS11 = {"StatusSTS":{"Time":"2019-02-08T16:58:49","Uptime":"0T00:11:32","Vcc":3.017,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":26,"Color":"420200","HSBColor":"2,100,26","Channel":[25,0,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Charles_Darwin","BSSId":"64:F6:9D:D5:E7:71","Channel":1,"RSSI":68}}}

```

ascillato2, 2019-02-09:
Hi,

Thanks for sharing all the information. Your config seems ok.

Can you try another power source for the nodemcu?

ascillato2, 2019-02-09:
Like one power supply for the led strip and another for your nodemcu but both with gnd connected between them

nromensky, 2019-02-09:
> Like one power supply for the led strip and another for your nodemcu but both with gnd connected between them

tried this, same result. everything works, but freezes with schemes

nromensky, 2019-02-11:
I think this is something to do with actual nodeMCU hardware. I configured sonoff basic with USE_WS2812_DMA, use RX pin. all works good, scheme works fine, and it does no freeze. I think we can close this problem

ascillato, 2019-02-11:
Great. Thanks for sharing. You have open this issue. Please close it.

closed 2019-02-11 by nromensky