arendst/Tasmota #5170: Please add function for sms and telephon functions with this sim900 Modul

dr-apple, 2019-02-08:
Hi please add sms and telephon functions with this sim900 Modul:

https://electronicfreakblog.wordpress.com/2014/04/27/das-sim900-gsmgprs-shield-und-der-leonardo/

It works with AT commands over Serial ;-)

ascillato2, 2019-02-08:
Hi,

The serial is already implemented in Tasmota. You can make rules you want to send AT commands.

See wiki for more infomation on commands and if you need further assistant please find us in the Tasmota Support Chat

closed 2019-02-08 by ascillato2

dr-apple, 2019-02-08:
No i will not works with rules ;-(

Please see the manual ;-)

ascillato2, 2019-02-08:
It works. I have done that using serial.

ascillato2, 2019-02-08:
You can send the AT commands using the Tasmota serialsend command. That serialsend can be issued by mqtt, by http and also inside a rule.
If you make rules, you can make a new event like `EVENT sendsms=light1ON` then in rules you can make
Rule1 1 
Rule1 on event#sendsms do backlog serialsend AT+; Delay 1; All the commands and then use the %value% variable to send it over serial.
Also you can store the phone number in a mem and call it from %mem1% etc

So, it can be done with rules. There is no problem with that. 

If you need further assistant please find us in the Tasmota Support Chat. Thanks

dr-apple, 2019-02-08:
Ok can you please tell me how i can see the Software Serial Bridge output from the device ?

ascillato2, 2019-02-08:
There is no need to use software serial. You can use hardware serial with command serialsend. Remember to put the seriallog to 0.

dr-apple, 2019-02-08:
I will it works with software serial ;-)

ascillato2, 2019-02-08:
The data sent from sim900 to Tasmota is going to be resent to your mqtt broker. You can use it in your home automation software until the request https://github.com/arendst/Sonoff-Tasmota/issues/4942 is done. I'm working on the trigger event for serial reception. Hope I could make a PR soon.

If you are in core 2.4.2 or 2.5.0 software serial will work the same as hardware serial. Just the commands change to sserialsend

ascillato2, 2019-02-08:
In core 2.3.0 you may experience some software watchdogs if using software serial

dr-apple, 2019-02-08:
:-( it dosn't work :
SSerialSend ATD+ +number;
SSerialSend2 ATD+ +number;
SSerialSend3 ATD+ +number;

Put i can see what the device respond :-(

What is wrong ?

Thank you for your help !

dr-apple, 2019-02-08:
I have tried it with the hardware serial port, seriallog 0 and then tried something to send, but nothing happens, and I also receive no messages from the sim900? What am I doing wrong ?

dr-apple, 2019-02-08:
yes the baudrate is 19200 set ;-)

ascillato2, 2019-02-09:
Hi, sorry I didn't answer before. At this moment I'm very busy finishing a project in my main job.

I have that module. It is very cool and useful. Give me some time and I will post here and in the wiki a full example using that module. I think this will be useful for other users too.

Thanks for the patience. :+1:

satekp, 2021-01-01:
Hi
   I'm trying to use sim800 mudule with Tasmota serial .
Could please share some example on  how to send on some trigger,  and power on action on  sms  received as trgger ?