arendst/Tasmota #5171: WiFi Smart Socket with RGB LED Indicator Energy Monitoring

Sergio-tix, 2019-02-08:
Hi!!
I have such a socket, Can you add it to the firmware?:
[image]
[image]

[image]

What I managed to find out:
[image]
This chip (HJL-01 1834CYH D79829IL) is very similar to HLW8012, Perhaps this is his complete equivalent.
[image]

Module ESP:
[image]
ESP8266EX, FM25F08. 
P.S. The device is extremely difficult to disassemble. Disassemble without damaging the case is impossible.

ascillato, 2019-02-08:
Great, thanks for the detailed description. It should be possible to use Tasmota on that device.

Please, save the original firmware with esptool.py as explained in the wiki and then erase all flash and finally flash it with the latest version of the precompiled bins. 

After the initial wifi config, you can set the module as generic and set the gpios as your description. 

Please, tell us how it goes. If you need further assistant, remember that it is available the Tasmota Support Chat.

If after that it works, we can add these pictures to the Tasmota wiki. 

Thanks

Jason2866, 2019-02-09:
Hi, in latest dev all needed GPIO pins can be selected in module generic.
Please be so kind and add -if you found it- the working config to wiki.
Thx!

arendst, 2019-02-09:
You might want to try the below configuration.

[image]

The power monitoring chip is compatible with BL0937 (not HLW8012).

You might need to experiment with GPIO12 being HLWBL SELi (inverted)

The leds might also be selected as just plain LED1 and LED2 (LED3 is not supported).

The relay might also be inverted so in that case select Relay1i

blakadder, 2019-02-09:
I added #5175 for the device since I own it myself and setting power monitoring in generic wasn't working quite as expected even after calibration. 
For flashing tuya-convert works fine and if you don't need the RGB light the Blitzwolf SHP module is good enough. button, power monitoring and relay work

closed 2019-02-09 by arendst

Sergio-tix, 2019-02-12:
> Возможно, вы захотите попробовать приведенную ниже конфигурацию.
> 
> [image]
> 
> Микросхема контроля мощности совместима с BL0937 (не HLW8012).
> 
> Возможно, вам придется поэкспериментировать с GPIO12, являющимся HLWBL SELi (инвертированным)
> 
> Светодиоды также могут быть выбраны как просто LED1 и LED2 (LED3 не поддерживается).
> 
> Реле также может быть инвертировано, поэтому в этом случае выберите Relay1i.

How did you install in GPIO BL0937 and HLWBL? I don't have this in the drop down menu.
P.S. Where can I get the version 6.4.1.**15** ? Is that all there?

blakadder, 2019-02-13:
[http://thehackbox.org/tasmota/](url)

marcobarriosfi, 2019-11-20:
> I added #5175 for the device since I own it myself and setting power monitoring in generic wasn't working quite as expected even after calibration.
> For flashing tuya-convert works fine and if you [...]

Hello, I believe I have the same device, but I cannot get the power monitoring to work. On Aliexpress it was called "TOP-MAX Smart Plug WiFi Socket 10A 2300W EU Outlet Remote Control Smart Timing Switch Works For Amazon Alexa/Google Assistant". Any help would be highly appreciated.

swiacek, 2019-12-09:
Configuration for plug without rgb leds

[image]

PabloCastellano, 2020-04-11:
> You might want to try the below configuration.
>
> [image]
>
> The power monitoring chip is compatible with BL0937 (not HLW8012).
>
> You might need to experiment with GPIO12 being HLWBL SELi (inverted)
>
> The [...]

I had to switch PWM2 and PWM3 otherwise the RGB colors were not correct

gerasimone, 2020-07-20:
Any chance to flash this smart plug without destroy the plug itself? Wireless flash works?
thanks

Jason2866, 2020-07-20:
@gerasimone  if the plug uses the TUYA Smartlife App you can try Tuya Convert.
If the Wifi MCU is still a ESP82XX it probably works.
You can look in template repo too https://templates.blakadder.com/

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

stosoorok, 2020-07-22:
Here is wrong template for this smart socket: https://templates.blakadder.com/avatto_OT08.html 
Save some time and use this AVATTO OT08 RGB smart socket template:
`{"NAME":"Avatto OT08","GPIO":[37,0,39,0,38,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}`

Got it from here: https://tasmota.github.io/docs/devices/RGB-Smart-Plug-16A/
And don't forget to calibrate: https://tasmota.github.io/docs/Power-Monitoring-Calibration/

Hope it helps someone. :)