arendst/Tasmota #5172: Last 2 Dev Firmwares cause Sonoff to go into a boot loop

DavidFW1960, 2019-02-08:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._
Last 2 dev builds, d509dcf and 78b2bd3 cause the sonoff (S22 and Sonoff SV) to go into a boot loop... every 8 seconds they reboot after an exception

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [ x] _Hardware used_ : S22, SonoffSV
- [ x] _Development/Compiler/Upload tools used_ :Visual Studio Code with PlatformIO IDE
- [ x] _If a pre-compiled release or development binary was used, which one?_ :n/a
- [ x] _You have tried latest release or development binaries?_ : Yes the precompiled for 2.4.2 core does not cause the issue
- [ x] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_
Using this my_user_config.h
```
// -- Master parameter control --------------------
#define CFG_HOLDER             4617              // [Reset 1] Change this value (max 32000) to load SECTION1 configuration parameters to flash

// -- Project -------------------------------------
#define PROJECT                "sonoff"          // PROJECT is used as the default topic delimiter

// If not selected the default will be SONOFF_BASIC
//#define MODULE                 SONOFF_BASIC      // [Module] Select default model from sonoff_template.h

#define SAVE_DATA              1                 // [SaveData] Save changed parameters to Flash (0 = disable, 1 - 3600 seconds)
#define SAVE_STATE             1                 // [SetOption0] Save changed power state to Flash (0 = disable, 1 = enable)
#define BOOT_LOOP_OFFSET       1                 // [SetOption36] Number of boot loops before starting restoring defaults (0 = disable, 1..200 = boot loops offset)

// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or enter a static IP address
#define WIFI_GATEWAY           "192.168.1.1"     // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.1.1"     // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "myssid"        // [Ssid1] Wifi SSID
#define STA_PASS1              "mypass"// [Password1] Wifi password
#define STA_SSID2              ""                // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""                // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_MANAGER      // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT, WIFI_SERIAL)
#define WIFI_CONFIG_NO_SSID    WIFI_MANAGER      // Default tool if wifi fails to connect and no SSID is configured
                                                 //   (WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_SERIAL)
                                                 //   *** NOTE: When WPS is disabled by USE_WPS below, WIFI_WPSCONFIG will execute WIFI_MANAGER ***
                                                 //   *** NOTE: When WIFI_MANAGER is disabled by USE_WEBSERVER below, WIFI_MANAGER will execute WIFI_SMARTCONFIG ***
                                                 //   *** NOTE: When WIFI_SMARTCONFIG is disabled by USE_SMARTCONFIG below, WIFI_SMARTCONFIG will execute WIFI_SERIAL ***

// -- Syslog --------------------------------------
#define SYS_LOG_HOST           ""                // [LogHost] (Linux) syslog host
#define SYS_LOG_PORT           514               // [LogPort] default syslog UDP port
#define SYS_LOG_LEVEL          LOG_LEVEL_NONE    // [SysLog] (LOG_LEVEL_NONE, LOG_LEVEL_ERROR, LOG_LEVEL_INFO, LOG_LEVEL_DEBUG, LOG_LEVEL_DEBUG_MORE)
#define SERIAL_LOG_LEVEL       LOG_LEVEL_INFO    // [SerialLog] (LOG_LEVEL_NONE, LOG_LEVEL_ERROR, LOG_LEVEL_INFO, LOG_LEVEL_DEBUG, LOG_LEVEL_DEBUG_MORE)
#define WEB_LOG_LEVEL          LOG_LEVEL_INFO    // [WebLog] (LOG_LEVEL_NONE, LOG_LEVEL_ERROR, LOG_LEVEL_INFO, LOG_LEVEL_DEBUG, LOG_LEVEL_DEBUG_MORE)

// -- Ota -----------------------------------------
#define OTA_URL                "http://thehackbox.org/tasmota/release/sonoff.bin"  // [OtaUrl]

// -- MQTT ----------------------------------------
#define MQTT_USE               1                 // [SetOption3] Select default MQTT use (0 = Off, 1 = On)

#define MQTT_HOST              "10.90.11.100"    // [MqttHost]
#define MQTT_FINGERPRINT1      "A5 02 FF 13 99 9F 8B 39 8E F1 83 4F 11 23 65 0B 32 36 FC 07"  // [MqttFingerprint1]
[... 910 lines cut ...]
I also compiled the current firmware with a my_user_config.h file that I used successfully with version 6ee3d1a

**Screenshots**
_If applicable, add screenshots to help explain your problem._
See below the console

**Additional context**
_Add any other context about the problem here._
```
00:00:00 Project sonoff sonoff3-0898 Version 6.4.1.15(sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 xxxxxx in mode 11N as sonoff3-0898...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on sonoff3-0898 with IP address 10.90.11.49
00:00:08 MQT: Attempting connection...
10:30:31 MQT: Connected
10:30:31 MQT: sonoff3/tele/LWT = Online (retained)
10:30:31 MQT: sonoff3/cmnd/POWER = 
10:30:31 MQT: sonoff3/tele/INFO1 = {"Module":"Sonoff Basic","Version":"6.4.1.15(sonoff)","FallbackTopic":"cmnd/DVES_A84382_fb/","GroupTopic":"sonoffs"}
10:30:31 MQT: sonoff3/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff3-0898","IPAddress":"10.90.11.49"}
10:30:31 MQT: sonoff3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x40232f2b epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
10:30:31 MQT: sonoff3/stat/RESULT = {"POWER":"ON"}
10:30:31 MQT: sonoff3/stat/POWER = ON (retained)
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-09:
Hi,

I couldn’t reproduce this issue. 

Your config seems ok. 

Please, could you try to wipe out the config settings using reset 5 command? 

If that can't be done or don't solve your issue, please do a full flash erase with esptool.py as explained in the wiki and then flash with esptool.py the latest precompiled bin from development branch.

DavidFW1960, 2019-02-09:
I did flash a precompiled firmware from dev ok but there is no precompiled 2.3.0 core and 2.4.2 and 2.5.0beta3 have been a disaster.

DavidFW1960, 2019-02-09:
It's tricky flashing when you only have 9 seconds before it crashes.... I flashed 7 sonoffs yesterday and managed to get them all back to my previous build and they have all been quite stable since.

I am just downloading the zip from dev and extracting, editing platformio.ini and user config and recompiling firmware. I have done this 50 times with no issue...

I think there is something specific wrong with this firmware I am compiling...

ascillato, 2019-02-09:
Ok. So the issue is not in the original Tasmota. Is in your building environment. 

The precompiled bins with core 2.3.0 are available for release only (6.4.1) check thehackbox.org page

The core 2.5.0 has been released 2 or 3 days ago and it is working great for me. Anyway, the selection of the core depends on your router brand. Some works better with 2.3.0 and others with 2.5.0

DavidFW1960, 2019-02-09:
So what should I check in my building environment? It hasn't previously been an issue.

I might try the released core 2.5.0 - is that in the latest dev releases?

DavidFW1960, 2019-02-09:
I just compiled my own with core 2.5.0 and it's not rebooting.... remains to be seen how stable it is with MQTT disconnects.
I don't even know where to begin looking to find troubles in my building environment - it's just Visual Studio Code with PlatformIO IDE..

ascillato2, 2019-02-09:
Good to know that now it is working for you.
Closing this issue as it isn't a Tasmota bug. 

For your building setup, you can try to delete all Tasmota files and also the core files and download them again. This I'm sure will solve your building setup issue because as you told me that you were downloading and copying, that could led to weird issues. In both, Tasmota and the core, some files were deleted and others renamed in the lasts versions, so, if you just overwrite files, the old files remain producing weird issues. 

If you need further assistant on that, please just ask here or in the Tasmota Support Chat. Thanks

closed 2019-02-09 by ascillato2