arendst/Tasmota #5173: Varn#State not working with ADDn

balajisraghavan, 2019-02-09:
On a sonoff pow r2, the `var1#state` rule does not trigger when `add1 n` is used to change the variable. I'm using the 20190208 23:01 GMT+2 build from thehackbox. For e..g the below rule does not fire when var1 is set to one by `add1` but works fine if `var1 1` is used.

`rule1
on system#boot do 
	VAR1 0
endon 
on ENERGY#Current>0.619 do
	VAR1 0
endon
on Time#Minute do 
	ADD1 1
endon  
on var1#STATE=1 do
	power off
endon`

ascillato, 2019-02-09:
@balajisraghavan

Please, update to latest Tasmota version from development branch (6.4.1.14)

This feature was recently added.

ascillato2, 2019-02-09:
The comparison elements for triggers has been changed recently. You need to use var1#STATE==1 instead of var1#STATE=1
Thanks

closed 2019-02-09 by ascillato2