arendst/Tasmota #5174: Gosund SP111: One has isolation on 230V, the other not.

heikoh81, 2019-02-09:
Hi everyone,

this issue is not related to the tasmota software, but to Gosund SP111 hardware.

Today I reveived a 2-pack Gosund SP111 from Amazon.de.
I opened up both plugs for soldering, and noticed, that hardware seems to be different.
While one has a black relay and one a white relay, what really worries me is that in one plug there is additonal isolation für the 230V male plugs entering the Gosund SP111. On the others, this isolaton is missing, so that 230V is very close to the capacitors.

I attached a picture of the problem.

Could that be a problem?

Greetings,
Heiko
[image]

Jason2866, 2019-02-09:
It is safer with this additional isolation. This means not that the other is dangerous at all...

ascillato2, 2019-02-09:
I totally agree with Jason. You should return the unsafe one.

closed 2019-02-09 by ascillato2

meingraham, 2019-02-09:
@heikoh81 - in the future, before cracking the device open (although in this case it was of interest in terms of understanding the devices's safety design), you might try [Tuya-Convert](https://github.com/ct-Open-Source/tuya-convert) to flash the device OTA w/o having to open it or solder any leads. If it doesn't succeed in flashing the device, there is no harm done and then you can crack the device open. It is **CONSIDERABLY** easier.

Regards.

Mike

arendst, 2019-02-09:
My latest SHP6 (=Gosund SP111) received last week from Banggood also did not have the extra isolation. What was worse one of the two screws connecting the PCA to the power pin had an intermittent connection. While I was hoping to use the Tuya-Convert procedure I needed to dismantle the whole unit and mount a new 1.5 millimeter longer screw I luckily had lying around. While getting frustrated about the inability to dismantle it easily I ruined the casing which is now held together with insulation tape...

In the end I did use the Tuya-Convert procedure while also having the serial interface active showing me what went wrong. The Tuya-Convert installation procedure needs the latest Node.js software which is not installed by the raspberry pi standard procedure ``apt-get update`` and ``apt-get upgrade``. I rather needed to download the latest ARM6 version from the Node.js website and install it.

Once done It updates fine.

meingraham, 2019-02-09:
@arendst 

> The Tuya-Convert installation procedure needs the latest Node.js software which is not installed by the raspberry pi standard procedure `apt-get update` and `apt-get upgrade`. I rather needed to download the latest ARM6 version from the Node.js website and install it.

Did you provide this feedback to Merlin Schumacher? It would be good information for him to add to the Tuya-Convert readme.

Thanks.

Mike

heikoh81, 2019-02-09:
Thanks for your replies. Will give Tuya-Convert a try.

@arendst:
Can confirm that dismantling and especially putting back together these plugs is very tricky.
Also found out that the 2 little screws seems to be non-magnetic, making it almost impossible to put them back where they belong. Don't know how they do this in their manufacturing process...

Jason2866, 2019-02-09:
I solder the needed wires without taking the PCB out of the case. 
Just loose the one screw to open the case

mcer12, 2021-05-05:
I've found this thread by accident and turns out I was using the one without insulion for a year! IT was seemingly working fine but it ABSOLUTELY isn't safe because the 230V can literally touch the low voltage capacitor. Also, bottom of the plastic case had a melted burn mark, I threw it out right away. Also the inner solder job seems sloppy at best. [image]

I also have the newer, 3450W version and it seems much better on the inside. Has the insulation and soldering seems higher quality. Don't buy the 2300W ones!