arendst/Tasmota #5176: SetOption21 1 doesn't show correct V when device not under load

blakadder, 2019-02-09:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
When setting up a power monitoring device under `Module 18` and calibrating the Voltage isn't showing the correct numbers when the device is not under load. It showed values from 0V to 8V. When I created a custom template for the device and used that everything works as expected-

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : Generic Tuya plug referenced in #5171 
- [x ] _Development/Compiler/Upload tools used_ : ArduinoIDE
- [ ] _If a pre-compiled release or development binary was used, which one?_ : 
- [x ] _You have tried latest release or development binaries?_ :
- [x ] _Provide the output of command_``status 0`` :

**To Reproduce**
Configure `Module 18` with power monitoring and enable `SetOption21`

**Expected behavior**
Voltage shown as a number near 220V

arendst, 2019-02-09:
Just tried on my SHP6 configured as

[image]

which provides Voltage when powered off as expected

[image]

So cannot replicate your situation.

ascillato2, 2019-02-12:
Hi,

Sorry, I can't reproduce your issue. Can you do a full erase with esptool.py as explained in the wiki and flash again with the precompiled bins from development branch. Then remember to calibrate it.

blakadder, 2019-02-13:
After configuring the device with a custom template `RGB Smart Plug` based on BlitzWolf SHP template everything works as expected.

closed 2019-02-13 by blakadder