arendst/Tasmota #5178: 433RFBRIDGE V2.0 - FLASH RF CHIP

SuoaJ, 2019-02-09:
Hi guys, 

I bought a Sonoff RF Bridge recently. The board states, "433RFBRIGE V2.0" (BRIGE is not my error, this is how it is written on the board). Anyway, I see information about R2 and R1 on the Internet. But mine does not say either R1 or R2. Is the "433RFBRIGE V2.0" the R1 version. I have already flashed the RF Bridge with Tasmota v6.4.1 but  I would also like to flash the RF chip. 

Question 1: Is flashing the RF chip still necessary with Tasmota v6.4.1? I want to make use of the beeper on the board?

Question 2: Is any hardware mod required to flash the RF chip on this particular board? 

Thanks.

Jason2866, 2019-02-09:
You can flash it as described in wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433

andrethomas2, 2019-02-09:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-09 by andrethomas2

SuoaJ, 2019-02-09:
I'm sorry, but I have spent the entire week trying to collect information on how to flash the RF chip but what is available on the blogs/wiki is just not clear. I've also read the link you sent multiple times. As I said earlier, I am already running Tasmota v6.4.1 on the RF Bridge. Q1: Can I simply upload the hex file to the chip through the web interface for this version of the RF bridge? I tried already and got this error, "Upload Failed. Failed to init RF chip". Or do I have to do the hardware mod which connects GPIO4 and C2Ck and GPIO5 and C2D? Q2: do I have to compile my own version of Tasmota to be able to flash the RF chip or will the v6.4.1 under releases work? Thanks again?

Jason2866, 2019-02-09:
You have to connect GPIO4 and C2Ck and GPIO5 and C2D.
After that upload efm firmware file via web frontend firmware upgrade

SuoaJ, 2019-02-09:
Great! It worked. Many Thanks, @Jason2866 and @andrethomas2 .

tyjtyj, 2024-05-08:
i visit this in 2024 and been trying for 6 hour.. why there is no RF433 R1 V2.0 on wiki page saying the tx and rx is reverse.

So there are 4 type for RF433 board
https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/

here are now 3 versions of the Sonoff RF Bridge 433: the R1 , R2 V1.0 and R2 V2.2.