arendst/Tasmota #5179: Difference between deleting flash with esptool and reset5

helmar74, 2019-02-09:
Just a short question from my side:

Is there a difference between deleting flash with
> esptool.py --port COM5 erase_flash

and the webconsole command:
< reset 5

Thanks a lot in advance

helmar

Jason2866, 2019-02-09:
Yes, esptool erases the whole flash. You need to flash Tasmota again
Reset 5 erases flash after Tasmota firmware. It emptys unused flash and resets Tasmota.
It is usefull when weired settings are done...

andrethomas, 2019-02-09:
Yes

`esptool.py --port COM5 erase_flash` erases the entire flash chip which requires reflashing of firmware as if it was never flashed before.

`reset 5` erases the section of flash not used by the firmware itself including the erasing of the sections used for rotational storage of settings. The only things that survive a `reset 5`  would be the firmware itself and your wifi credentials for ssid1 and ssid2.

andrethomas2, 2019-02-09:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-09 by andrethomas2

helmar74, 2019-02-09:
Great thanks a lot for your help!