arendst/Tasmota #5181: tasmota device only connects to MQTTserver when "sudo mosquito" is run on MQTT server

frankeng, 2019-02-09:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino IDE, board 2.5.0
- [X] Hardware used :  T1
- [ ] If a pre-compiled release or development binary was used, which one? :
 Compiled source code
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` 
```
MQT: Connect failed to 192.168.1.14:1883, rc -2. Retry in 10 sec
```
I have 2 T1 sonoff devices .
One of my T1 with tasmota is connecting to the MQtt server, the other one first give the error:
MQT: Connect failed to 192.168.1.14:1883, rc 5. Retry in 10 sec

After restart of the MQTT server with:
`sudo systemctl disable mosquitto`
`sudo systemctl enable mosquitto`
Both devices give the following error:
`MQT: Connect failed to 192.168.1.14:1883, rc -2. Retry in 10 sec`

when I then type in the command "sudo mosquitto" on the server terminal.
both devices are working fine.
When I do "ctrl+c" in the smae server terminal, the error rc=-2 comes back for both devices.

settings of the device:
**device1:**
MQTT Host	192.168.1.14
MQTT Port	1883
MQTT User	openhab
MQTT Client	rolluikebbadkamer
MQTT Topic	bathR/window/shutter
MQTT Group Topic	sonoffs
MQTT Full Topic	cmnd/bathR/window/shutter/
MQTT Fallback Topic	cmnd/rolluikenouderkamer_fb/

**device2:**
MQTT Host	192.168.1.14
MQTT Port	1883
MQTT User	openhab
MQTT Client	rolluikenouderkamer
MQTT Topic	parentsR/window/shutter
MQTT Group Topic	sonoffs
MQTT Full Topic	cmnd/parentsR/window/shutter/
MQTT Fallback Topic	cmnd/rolluikenouderkamer_fb/

Can anyone tell me what is happening

ascillato, 2019-02-09:
Hi,

Sorry, that is not a Tasmota issue. You should ask in Openhab site.

Seems that your mosquitto installation does not have write privileges, but will be better to ask that in Openhab. Thanks

closed 2019-02-09 by ascillato2

frankeng, 2019-02-09:
But all other ESP and arduino device are working correct  with mosquitto.
I installed openhab and mosquito individualliy on the same server.
If  I stop openhab, I still got the same error. while mosquitto is stil running.
So why should this be an openhab issue?

ascillato, 2019-02-09:
No one said that it is an OpenHab issue. It is a mosquitto issue. Seems that your mosquitto installation does not have write privileges. So, it is better to ask in Openhab site, how to install mosquitto correctly.

Sorry, it is not a Tasmota software issue.

frankeng, 2019-02-11:
Solved see:
https://community.openhab.org/t/solved-mqtt-sonoff-device-cannot-connect-to-broker-rc-5/67264/21