arendst/Tasmota #5182: Best configuration for LED control and motion sensor?

poldim, 2019-02-10:
I've got a motion sensor and an LED strip connected to an ESP.  I expected that having the LED driven by PWM**1** and the PIR on Switch**2**, that they would operate individually from one another.  However, in practice, it seems that the PIR sensor is **always** driving the PWM.

What is the best configuration for this setup?  And what is the best configuration for extending this setup to 4 channels for RGBW LEDs instead of my single color LED?

Hardware details:

```
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Kitchen Undercab LED"],"Topic":"kitchen_undercab_led","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"kitchen_undercab_led_motion","SwitchMode":[0,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"Software/System restart","Uptime":"2T01:01:56","StartupUTC":"2019-02-07T19:56:46","Sleep":0,"BootCount":43,"SaveCount":1754,"SaveAddress":"F9000"}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.14(ee2660b-sonoff)","BuildDateTime":"2019-02-05T19:12:15","Boot":31,"Core":"2_5_0_BETA3","SDK":"3.0.0-dev(c0f7b44)"}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HAN",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00128009","280500000100000000000000000000000000","00000200"]}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS4 = {"StatusMEM":{"ProgramSize":552,"Free":448,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS5 = {"StatusNET":{"Hostname":"TASMOTA_kitchen_undercab_led","IPAddress":"192.168.3.154","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"CC:50:E3:08:13:B8","Webserver":2,"WifiConfig":4}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"kitchen_undercab_led","MqttClient":"kitchen_undercab_led","MqttUser":"poldim","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Feb 09 20:58:42 2019","Local":"Sat Feb 09 21:58:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:08","Sunset":"18:00"}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS10 = {"StatusSNS":{"Time":"2019-02-09T21:58:42","Switch2":"OFF"}}
21:58:42 MQT: homeassistant/light/kitchen_undercab_led/stat/STATUS11 = {"StatusSTS":{"Time":"2019-02-09T21:58:42","Uptime":"2T01:01:56","Vcc":3.024,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"OFF","Dimmer":100,"Fade":"ON","Speed":4,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"HAN","BSSId":"06:18:D6:F5:AD:3D","Channel":11,"RSSI":64}}}```

I've got the following hardware:
![image](https://user-images.githubusercontent.com/16869713/52528430-49b62800-2c93-11e9-83d3-8fd14c618189.png)

Tasmota is configured with 'GPIO13: 37 (PWM1)' and 'GPIO14 : 10 (Switch2)' [pins 7 and 5 on the 
WEMOS respectively].
![image](https://user-images.githubusercontent.com/16869713/52528454-e082e480-2c93-11e9-8df6-013a640a6d68.png)

ascillato2, 2019-02-10:
Hi,

This is a multi duplicated issue. 

That is a feature. 

Solution: use rules to make Tasmota do the behaviour you want. 

Please, see the examples of the wiki for this case like: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

Thanks

closed 2019-02-10 by ascillato2

poldim, 2019-02-10:
Thanks,

I'm reading through this to make sure I understand correctly.  It looks like I should be implementing the second example shown, only for the single press.  However, the button [Action Matrix](https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices#action-matrix) seems to conflict with the desired end result.  It's saying that it will toggle the relay in the example's 0-1-0 config (which equals the table's 1-0-0 row).  

Am I reading this correctly?

[image]

ascillato, 2019-02-10:
For PIR use switch. Not button.

ascillato, 2019-02-10:
Please, see examples in rules