arendst/Tasmota #5185: Help with Supernight dual wifi plug

DavinKD, 2019-02-10:
I've been able to flash the following plug.

https://smile.amazon.com/gp/product/B07K1QSFFJ/ref=ppx_yo_dt_b_asin_title_o01__o00_s00?ie=UTF8&psc=1

Running the latest build of Tasmota.

I've been able to figure out the following GPIO pins

GPIO1 = Left Socket Button
GPIO3 = Left Socket Relay
GPIO12 = Right Socket Relay
GPIO14 = Red LED for both buttons

For the life of me, I can't  figure out which pin is the the right socket button.  I've tried all the pins from 0 to 17.

I think the "night light" is directly tied to a light sensor and may not be controllable at all.  It looks to be a pain to tear open, so I'm trying to avoid that.  It also has energy monitoring, but I'll cross that bridge once I at least get the basics going.

Any advice?

DavinKD, 2019-02-10:
Just a quick update.  I've still not figured out the second button, but I do have energy monitoring working now.

My current pinout is...
GPIO1 = Left Socket Button
GPIO3 = Left Socket Relay
GPIO4 = HLW8012/Voltage
GPIO5 = HLW8012 CF Power
GPIO12 = Right Socket Relay
GPIO13 = HLW8012 Sel Output 
GPIO14 = Red LED for both buttons

Pins 6,7,8 and 11 cause hard resets when I try to set them to switches or buttons.

ascillato2, 2019-02-10:
Seems that you will need to open it to do some reverse engineering in order to know exactly what pins are connected and to what.
Please, then add this information to the wiki under other devices. Will be super useful for other users. 
Thanks a lot for sharing this information. :+1:

If you need further assistant please find us in the Tasmota Support Chat 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-10 by ascillato2

DavinKD, 2019-02-12:
I opened it up and now I'm wishing I had verified the functionality of the second button before flashing.  When I trace the path of the circuit, it's purposely notched out so that the second button is never connected to the ESP chip.  I had to destroy the plug to figure this out.

I don't think I'll buy any more just to figure that out.  I don't like the always on night light anyways.

not5, 2019-03-10:
How did you open the device up? Anyway to open without destroying?

Also, can you post pics of the interior? Any extra pins available?

DavinKD, 2019-03-10:
I destroyed it, I still have the boards, so I'll snap some pictures.

not5, 2019-03-10:
Could you tell if it was just glued together or there were tabs, etc? I appreciate it!

DavinKD, 2019-03-10:
It was glued up pretty tight. Probably ultrasonic welded imo. No tabs.

fiercedeitylink, 2019-03-13:
I bought two of these and can confirm that the right button does indeed control the right socket on the stock firmware. I haven't figured out how to get the right button to work with Tasmota either, though.

DavinKD, 2019-03-13:
Good to know. I still owe you guys some pictures.

DavinKD, 2019-03-19:
Finally got around to adding some pictures.  See here.

https://github.com/DavinKD/SmartThings/tree/master/DeviceImages/SuperNight

One thing you'll notice is that on the board that has the buttons, there's a notch that cuts right through the trace that comes from the non-working button.

not5, 2019-03-19:
Thanks for posting the pics! Looks like the male outlet portion basically disintegrated when you attempted disassembling.

I'm wondering if there's a better way to separate...

DavinKD, 2019-03-19:
Things tend to disintegrate when a dremel with a cutting disk is involved.  Wish I had a good macro lens camera to get some better shots.  I believe this thing should have energy monitoring too, but since I couldn't get the second button to work, nor could I disable the night light, I gave up.  The SP201 based plugs worked very well for me.

not5, 2019-04-02:
I was able to crack these open without destroying them. You have to slip a tool in the seam on the back and twist out. See [image](https://i.imgur.com/xuaPbYI.jpg).

gongtao0607, 2019-04-18:
I was able to track down the circuit. The right button is connected to ADC0.
More specifically, when pressed, ADC0 reads 1024. Otherwise, 0.
To use right button, I have to disable USE_ADC_VCC (I flashed sonoff-sensor.bin)
and add a rule:
`Rule1 on Analog#A0div10>50 do Power2 2 endon`
The rule triggering makes two button feel different, but it works!

gongtao0607, 2019-04-18:
Also to disable the nigh light, I short the photodiode. (Right at the center of the Top PCB. Once you crack the case open, it's quite easy to solder bridge it.)

DavinKD, 2019-04-18:
Excellent work. I'll try that on my one good plug tomorrow. I don't feel like opening to disable the night light, but at least the button will work.

not5, 2019-04-18:
Are there any open pins that we can use to enable/disable night light?

gongtao0607, 2019-04-18:
I believe GPIO0 and GPIO2 are open as testpins, on TYWE2S, although the pad is a little tiny and no solder paste was applied. I haven't try but I think it's something can be achieved.

DavinKD, 2019-04-18:
I've used the sonoff-sensor.bin and entered that rule.  It does work, but it's a bit "touchy".  Nice work, though.

meingraham, 2019-04-18:
@DavinKD **_OFF TOPIC_**- please take a look at [KS-604S wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Kesen-KS-604S). Confirm that you flashed a [KS-604S](https://blakadder.github.io/templates/kesen_KS-604S.html) (with USB) and not a [KS-604](https://blakadder.github.io/templates/kesen_KS-604.html) (no USB). My GPIO configuration was different and also you do not mention any setup for the USB port. Thanks

arendst, 2019-04-18:
I'm working on an option where the ADC0 input will act as just another button. If that's done then you should have a better response when the button is clocked.

DavinKD, 2019-04-18:
@meingraham , for sure I have the USB one.  I'm at work and don't have access to it right now, but I can take a look later.  Is there a relay for the USB?

meingraham, 2019-04-18:
@DavinKD - Are you on Discord? Let's take this dialog over there. Would like to document the variations to best document the KS-604S configuration(s).

Yes, my KS-604S devices have a means to enable and disable the USB port. It's not a mechanical relay as far as I can tell because I can turn the port on and off yet there is no relay "click".

gongtao0607, 2019-04-18:
@not5 I've just traced the night light circuit board. The automatic night light runs a separate power directly from N and L, using a diode bridge. The ESP8266 runs from a single diode rectifier. As they use different grounds, connect them requires a optocoupler, and also a driver at emitter side.
TLDR: not as simple as a jumper wire.

not5, 2019-04-18:
@gongtao0607 maybe it's a matter of replacing light sensor with 2Mohm resistor and having a relay switch power to LED board?

gongtao0607, 2019-04-19:
@not5 Yes you can always wire a relay switch. But then you will need a separate circuit drive it from GPIO. I don't think a relay board will fit inside the case, thinking about some wires have Live-line voltage. Optocoupler may be a better idea if you really need to control those LEDs.