arendst/Tasmota #5187: Blitzwolf SHP6/Gosund SP111: Where can I find binary for development firmware >=6.4.1.7

heikoh81, 2019-02-10:
Hi,

I want to flash Gosund SP111 that is identical to Blitzwolf SHP6.
Wiki states:
> Attention: Please use development firmware >=6.4.1.7

Where can I find binaries of the development firmware.
I tried the development branch on Github, but on clicking the releases-link, I'm back t master branch.

Can't compile it myself, any help appreciated.

Greetings,
Heiko

Jason2866, 2019-02-10:
Reading main page of Tasmota Github would answer ;-)
https://github.com/arendst/Sonoff-Tasmota
[image]
http://thehackbox.org/tasmota/

ascillato2, 2019-02-10:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-10 by ascillato2

heikoh81, 2019-02-10:
@Jason2866 
Thanks. I had visited that page before, but was irritated as there are no version numbers.
I was looking for version 6.4.17.

Jason2866, 2019-02-10:
Thats why `>`