arendst/Tasmota #5188: Sonoff RF Bridge - RF -> Http Requests

Electrobub, 2019-02-10:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Simple functionality for the RF Bridge that allows to request HTTP (after receiving an RF command) 

**Describe the solution you'd like**
Functionality that allows opening HTTP requests. Especially useful for the RF Bridge. For eg. upon receiving an RF command it can then open a HTTP request (as opposed to only sending a msg on MQTT) This would allow the interconnection of RF remotes to control WiFi based switches (a lamp on a Sonoff basic module toggled by http), without having to install an MQTT based system.

**Describe alternatives you've considered**
(Installing MQTT on a server/PI with software/scripts that would handle the RF -> WiFi signal) - as usual

**Additional context**
This would give a very lightweight solution, with reduced components, for those that wish to do very basic switching or more, without the need of  having to setup a home automation backbone)
Allowing the RF Bridge to provide a very simplistic 'all-in-one' solution.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-10:
Hi,

It is already implemented.

See commands in the wiki for syntax

Also, if you want to receive a RF code and produce a HTTP command, you can use RULES feature and WEBSEND command

If you need further assistant on this, please, find us in the Tasmota support Chat.

Thanks.

ascillato2, 2019-02-10:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-10 by ascillato2

Electrobub, 2019-02-10:
Thank you for listing the commands. I had missed them. I will give it a go!