arendst/Tasmota #5190: Rule triggered and activating Websend - Unreliably switching basic module

Electrobub, 2019-02-10:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ]x Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used :
PlatformIO / FTDI
- [x ] Hardware used : 
Sonoff RF Bridge 433
- [x ] If a pre-compiled release or development binary was used, which one? :
- [x ] You have tried latest release or development binaries? :
- [ x] Provide the output of command ``status 0`` :
```
19:27:03 RSL: STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:27:03 RSL: STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:20","StartupUTC":"2019-02-10T18:26:43","Sleep":50,"BootCount":56,"SaveCount":108,"SaveAddress":"FA000"}}
19:27:03 RSL: STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
19:27:03 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Box14512",""],"TelePeriod":300,"SetOption":["00008001","558180C0","00000000"]}}
19:27:03 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
19:27:03 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-7698","IPAddress":"192.168.178.24","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"61:01:92:BE:6E:13","Webserver":2,"WifiConfig":4}}
19:27:03 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 10 18:27:03 2019","Local":"Sun Feb 10 19:27:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:07","Sunset":"18:01"}}
19:27:03 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-02-10T19:27:03"}}
19:27:03 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-02-10T19:27:03","Uptime":"0T00:00:20","Vcc":3.504,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Box14512","BSSId":"F0:A0:24:36:AE:43","Channel":1,"RSSI":100}}}
```

Hello, I am using the RF Bridge 433, with the intent of having it use a HTTP request to switch on/off a Sonoff basic module.

I have implemented it using Rules (from the good advice that I got from here). I have tested triggering it and works! Sometimes! It seems to work every now but is quite unpredictable?? (I have no problem turning on/off the switch quickly through the WebUI and HTTP requests in my browser)

To create the rule:
```
rule1 on event#temp>50 do  WebSend [192.168.178.25:80] Power TOGGLE endon
rule1 1
```
Triggering with
`event temp=60`

```
CMD: event temp=60
RSL: RESULT = {"Event":"Done"}
RUL: EVENT#TEMP>50 performs "WebSend [192.168.178.25:80] Power TOGGLE"
RSL: RESULT = {"WebSend":"Done"}
```

It toggles at times but not most of the time... (irregardless if left alone for a while or not)

Has anyone encountered this? Any tips?

Electrobub, 2019-02-10:
I have tried sending a few commands a bit closer together. On the rare occasion it will switch as well, about 15 - 20 seconds after the last cmd was sent... But this does not explain when a single command is sent individually out of the blue and it doesn't switch

andrethomas, 2019-02-10:
Sorry, I cannot reproduce - it works flawlessly for me.

Electrobub, 2019-02-10:
I reset the device, reloaded firmware from: http://thehackbox.org/tasmota/release/sonoff.bin
(Did some basic setting changes: Turn off MQTT, Change module type to RF Bridge)
And copied the above 2 lines for the rules.

And I still get intermittent results... (Roughly 1 out of 5 times it will work)

andrethomas, 2019-02-10:
Can only be related to your network setup maybe.

To test further I set the event from node-red to rule out intermittent problems but as of now its been clicking a relay (toggle) for the 30 minutes.

I count the stat messages from the target device on node-red also and they match so not a single relay change was dropped.

Only other difference is that I am using the development binary but that should not be the cause... don't recall anything to websend being changed recently.

ascillato, 2019-02-11:
Your issue seems to be wifi inestability. Please, use command weblog 4 to see more debug information in the console.

ascillato2, 2019-02-11:
You are using core 2.4.2

Depending in your router brand, some core versions works better than others. 

Please, check the following link to select another core and test which one works better for you: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

If you need further assistant please find us in the Tasmota Support Chat. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-11 by ascillato2

Electrobub, 2019-02-11:
Thank you for the responses. Got it working!

I found that the router I am using has been listed as having problems with core 2.4.2
I upgraded to Core 2.5.0 and it is working perfectly! Hasn't skipped a beat!

Thank you for the responses, I hope this info can help others too