arendst/Tasmota #5192: MQTT does not work for Tasmota Generic module

TamojitSaha, 2019-02-11:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [x] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` :
```
09:53:48 RSL: stat/mysonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"mysonoff","ButtonTopic":"0","Power":60,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:53:48 RSL: stat/mysonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:26:52","StartupUTC":"2019-02-11T08:26:56","Sleep":50,"BootCount":45,"SaveCount":222,"SaveAddress":"3F6000"}}
09:53:48 RSL: stat/mysonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.14(display)","BuildDateTime":"2019-02-10T20:05:42","Boot":5,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
09:53:48 RSL: stat/mysonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Tamojit","Tapan & Tamojit"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
09:53:48 RSL: stat/mysonoff/STATUS4 = {"StatusMEM":{"ProgramSize":487,"Free":2584,"Heap":10,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","0F20048C","00000FA0","00000092","000000C0"]}}
09:53:48 RSL: stat/mysonoff/STATUS5 = {"StatusNET":{"Hostname":"mysonoff-1642","IPAddress":"192.168.43.132","Gateway":"192.168.43.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.43.1","Mac":"60:01:94:1F:86:6A","Webserver":2,"WifiConfig":2}}
09:53:48 RSL: stat/mysonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.43.44","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1F866A","MqttUser":"pi","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:53:48 RSL: stat/mysonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 11 08:53:48 2019","Local":"Mon Feb 11 09:53:48 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:05","Sunset":"18:03"}}
09:53:48 RSL: stat/mysonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-11T09:53:48","DHT11":{"Temperature":null,"Humidity":null},"TempUnit":"C"}}
09:53:48 RSL: stat/mysonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-11T09:53:48","Uptime":"0T00:26:52","Vcc":2.922,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"ON","POWER5":"ON","POWER6":"ON","Dimmer":57,"Fade":"OFF","Speed":1,"LedTable":"OFF","POWER7":"OFF","Wifi":{"AP":1,"SSId":"Tamojit","BSSId":"64:CC:2E:8F:95:43","Channel":11,"RSSI":100}}}
```
I am running tasmota in *NodeMCU* module and configured it as a **Generic** module. MQTT broker is configured in Raspberry Pi.  
MQTT publishing, subscription and web commands works when the module was configured as **Sonoff 4CH** but it does not when changed to **Generic**
**Issues**
1. Web commands does not work for generic, i.e no JSON output is returned
2. Cannot subscribe or publish to any topic

Please help me!

Jason2866, 2019-02-11:
Please try with a precompiled Tasmota version. 
http://thehackbox.org/tasmota/020500/sonoff-display.bin
Something is wrong with your selfcompiled version.
You used core 2.4.0 which has manny issues and isnt supported from Tasmota side

TamojitSaha, 2019-02-11:
@Jason2866 Nothings happens after I upload the binary. Prints out gibberish...

ascillato, 2019-02-11:
Hi,

Sorry, I can't reproduce your issue. It is working fine for me with generic. 

Seems that you are having an issue with your mqtt broker. User and password are correct? Yout mqtt broker has write access?

ascillato, 2019-02-11:
If you continue having issues, may be your flash has corrupt data in the flash. You can do a full flash erase with esptool.py as explained in the wiki and flash again with the precompiled bins.

ascillato2, 2019-02-11:
Have you checked that your mqtt credentials are ok?

dezral, 2019-02-11:
im using the current version on esp-01 generic and mqtt works fine on precompiled bin.

ascillato2, 2019-02-12:
Hi,

Any news on this?

TamojitSaha, 2019-02-12:
Hello,
Appologies for late reply. 
Solved the issue after flashing ESP8266 with new firmware from Hackbox.
Had some doubts..

- Running `webserver xxxxx` in comand bar of console window displays the password in plain text and even after relogging the password can be seen in plain text. 
The proposed solution is to ask for re-entering webserver password within certain timeout and if the user fails to renter the correct, the password is discarded.
After successful re-entering correct password, a confirmation is show that password was accepted.
Can this be achieved?
If yes how? I would be happy to dig inside the firmware...

ascillato2, 2019-02-12:
Hi, great to know that it is working now for you.

ascillato2, 2019-02-12:
About the password, I don't quite understand what you mean. Are you setting the password by commands? In that case, is a command, and will not be ofuscated.
If you enter the password by the web ui. It is already ofuscated and it is never sent in plain text to the browser. 
If the browser ask for password, it is being asked using the log in window of your web browser

closed 2019-02-12 by ascillato2

TamojitSaha, 2019-02-12:
Ok, lemme breakdown the process I wanted to ask.

-  After a fresh flashing of tasmota firmware, you connect your device to the network and it works as it should.

- Then navigate to the console window(in browser). In the command bar, you type `webpassword [your password]` and hit enter. Immediately the browser prompts for login details. 
You put the correct details and log in.
In the same console log area, the input password is shown as plain text.

- My question was whether it is possible to hide the web-ui password in the log after it isset by the command `webpassword xxxxx`?

ascillato, 2019-02-12:
Ah. Ok.

If you use it as a command is going to be like that. 

If you don't want to see that, you should go configuration menu, hit extra config and put there the password.