arendst/Tasmota #5193: LCD2004 DisplayMode 1 the time is clipped the last gidit

wongnam, 2019-02-11:
**Describe the bug**
_LCD2004 DisplayMode 1 the TASMOTA time is clipped the last gidit, see screenshot_

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ :  NodeMCU, LCD2004 I2C
- [x] _Development/Compiler/Upload tools used_ : VSC, Tasmosta latest verion 6.4.1.15 core 2.4.2
- [x] _If a pre-compiled release or development binary was used, which one?_ : Self complied
- [x] _You have tried latest release or development binaries?_ : Y
- [x] _Provide the output of command_``status 0`` :
```
00:00:00 Project sonoff Sonoff Version 6.4.1.15(sonoff)-2_4_2
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP1 iot-2 in mode 11N as sonoff-2372...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-2372 with IP address 192.168.12.120
20:45:19 MQT: Attempting connection...
20:45:19 MQT: Connected
20:45:19 MQT: tele/sonoff/LWT = Online (retained)
20:45:19 MQT: cmnd/sonoff/POWER = 
20:45:19 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"6.4.1.15(sonoff)","FallbackTopic":"cmnd/DVES_438944_fb/","GroupTopic":"sonoffs"}
20:45:19 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-2372","IPAddress":"192.168.12.120"}
20:45:19 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
20:45:19 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
20:45:19 MQT: stat/sonoff/POWER = ON
20:45:27 MQT: tele/sonoff/STATE = {"Time":"2019-02-11T20:45:27","Uptime":"0T00:00:14","Vcc":3.018,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":76}}
20:45:27 MQT: tele/sonoff/SENSOR = {"Time":"2019-02-11T20:45:27","ENERGY":{"TotalStartTime":"2019-02-11T20:34:06","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
20:46:01 CMD: status 0
20:46:01 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:46:01 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:48","StartupUTC":"2019-02-11T13:45:13","Sleep":50,"BootCount":5,"SaveCount":16,"SaveAddress":"F4000"}}
20:46:01 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.15(sonoff)","BuildDateTime":"2019-02-11T19:48:26","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
20:46:01 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-2","Wong"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
20:46:01 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FAAE794","240783A0","23B617CE","00003BC0"]}}
20:46:01 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2372","IPAddress":"192.168.12.120","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"2C:3A:E8:43:89:44","Webserver":2,"WifiConfig":4}}
20:46:01 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_438944","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:46:01 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 11 13:46:01 2019","Local":"Mon Feb 11 20:46:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00","Sunrise":"06:14","Sunset":"17:59"}}
20:46:01 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:46:01 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-11T20:46:01","ENERGY":{"TotalStartTime":"2019-02-11T20:34:06","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
20:46:01 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-11T20:46:01","Uptime":"0T00:00:48","Vcc":2.997,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":84}}}
```

**To Reproduce**
_Steps to reproduce the behavior: DisplayMode 1 on LCD2004_

**Expected behavior**
It should display full time and date without clipped.

**Screenshots**
[image]

arendst, 2019-02-11:
Pls change line 110 in xdsp_01_lcd.ino from
```
  strlcpy(line +offset, txt, len);
```
in to
```
  strlcpy(line +offset, txt, len +1);
```
and report back the result

wongnam, 2019-02-11:
@arendst Fixed. Thank you.

closed 2019-02-11 by arendst

wongnam, 2019-02-11:
Update: but in DisplayMode 3 and 5 while its scroll up that the information has sticky letters issue, because my LCD is 4 lines.

[image]