arendst/Tasmota #5194: Add support for NRF24L01 and LT8900

qingz2004, 2019-02-11:
**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
Need a bridge to support 2.4G RF communication used by MiLight and other product.

**Describe the solution you'd like**
Add SPI for NRF24L01 or LT8900.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
https://github.com/sidoh/esp8266_milight_hub

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-02-11:
Why?
You can use the esp8266_milight_hub to communicate to Milight.
Since the esp8266_milight_hub uses mqtt too, no hassle to use all in a Home Automation system.

qingz2004, 2019-02-11:
I'm currently using that project, but I'd like to use the spare IOs from the same NodeMCU for something else. And I prefer to use the same Tasmota firmware for all of my esp8266 devices.
I'm looking for the same solution as Sonoff 433M RF bridge.

blakadder, 2019-02-11:
look at openmqttgateway then

closed 2019-02-11 by ascillato2