arendst/Tasmota #5197: second button on Sonoff T1 2ch not working (sonoff.bin)

joggs, 2019-02-11:
Second button on Sonoff T1 2ch touch panel is not working using the dev builds of sonoff.bin (as of 190211) for 2.4.2 and 2.5.0 , 6.4.1.
The release version of 6.4.1 works well.
the second button works well from the web ui, but not if you push the button on the panel.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : Sonoff T1 2ch
- [ ] _Development/Compiler/Upload tools used_ : webui
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [x ] _You have tried latest release or development binaries?_ : release works, deve does not
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_
install devbuilds of sonoff.bin for 2.4.2 or 2.5.0 and the second button does not work if you push the touchplate. It works well in the web ui.

ascillato, 2019-02-11:
Hi,

Can you confirm if that button was working with the original firmware? 

Can you confirm if there is no wire loose inside the module? 

You can also try to check if the gpios are the same as the template for that module. Sometimes happens that a device is not original and it does not have the same pinout.

Also remember that for support it is available the Tasmota Support Chat for live Support.

wongnam, 2019-02-12:
My Sonoff T1 2CH is working normally on 6.4.1.15 core 2.4.2
Make sure that you are selecting Sonoff T1 2CH (29) in Module parameters.

tompipe, 2019-02-12:
Ensure SetOption51 is enabled

joggs, 2019-02-12:
@ascillato yes, the button worked with the original fw and also with the current release version. No wire is loose. Also, the button still works if i click it from the web gui. It is just the physical button on the touch panel that stopped working after the update.

I actually have 2 T1 2ch and they both stopped working after the fw update. I also updated a T1 1ch and that works.

@wongnam Yes Sonoff T1 2CH (29) is selected

@tompipe I will try option 51 after work today. How will that change affect the T1 ? The explanation says "Enable to view extra pins for ESP8285 in Module", but I do not understand it in this case :-)

ascillato2, 2019-02-12:
Please, try the new version from today from the development branch. Thanks

joggs, 2019-02-12:
The same problem with 20190212 - http://thehackbox.org/tasmota/020500/sonoff.bin

ascillato2, 2019-02-12:
Have you erased all flash with esptool.py as explained in the wiki and then flash the firmware?

joggs, 2019-02-12:
yes. i have tried the first time on web interface and after that with esptool.py including erase

joggs, 2019-02-12:
Strange indeed. The 2.3.0 version is the most stable for me, also no connection drops. I can always roll back to that one, but I am curious to see if the 2.5 version has a more stable wifi connection.
Strange if it is working for other people. 
Here is my status 0
20:02:44 RSL: stat/sonoff/STATUS = {"Status":{"Module":29,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:02:44 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:08:04","StartupUTC":"2019-02-12T18:54:40","Sleep":50,"BootCount":12,"SaveCount":66,"SaveAddress":"FA000"}}
20:02:44 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.16(e779c7b-sonoff)","BuildDateTime":"2019-02-12T17:12:11","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
20:02:44 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dd-wrt12",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
20:02:44 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":556,"Free":444,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000783A0","23B617CE","00003BC0"]}}
20:02:44 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3650","IPAddress":"192.168.0.143","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"","Webserver":2,"WifiConfig":1}}
20:02:44 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DEAE4242","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:02:44 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 12 19:02:44 2019","Local":"Tue Feb 12 20:02:44 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:03","Sunset":"18:05"}}
20:02:44 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-12T20:02:44"}}
20:02:44 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-12T20:02:44","Uptime":"0T00:08:04","Vcc":4.466,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"dd-wrt12","BSSId":"44:D9:E7:91:95:97","Channel":8,"RSSI":100}}}

ascillato2, 2019-02-12:
>Could there be a different hardware revision so that a matching template is not found?

The internal Tasmota template of gpios config for that device was not changed.

wongnam, 2019-02-12:
@joggs Use serial to erase the flash with esptool.py and then flash it diect from the latest fw with core 2.4.2. You will see the different.
Note: as i have experienced that don't flash 2.3.0 and upgrade to 2.4.2, to do so i am not sure it will stable with some hw.

joggs, 2019-02-13:
I have tried all different versions now and performed eraseflash + serial flash with all mentioned tools and I still have the same problem with the second button that seems to have appeared somewhere after the the release of the current release version and I get this on 2 different T1 2ch units.
I have rolled back to 2.30 and that version works perfectly, so I will continue to use that version.
Strange indeed. I am not new to flashing and firmware tweaking so I am pretty certain that I have done everything correctly.
There must be a change in the code between the release version and current version that broke the functionality for my units/revisions.

Jason2866, 2019-02-14:
It has to be Arduino Core related if it works with Core 2.3.0.

> I have rolled back to 2.30 and that version works perfectly, so I will continue to use that version.

Tasmota "hooks up" on the Core. There is no general change in code of Tasmota if you use Core 2.3.0, 2.4.2 or 2.5.0 (only a few conditional bug fixes for the core...)

wongnam, 2019-02-15:
@joggs I can replicate your T1-US 2C issue with latest develop version 6.4.1.17 core 2.4.2. The 2nd touch button no working. the soft buttons in Web-UI are still working well.
I can confirm that the 6.4.1.15 is working well for T1-US 2C
@ascillato please consider this case is an issue and have an investigate. Thanks.

Jason2866, 2019-02-15:
Is it verified that all buttons work with core 2.3.0 with latest dev version?
Or is the issue with all core versions?

yellowonblack, 2019-02-15:
Yesterday, I updated a Sonoff T1 2CH Module and I can confirm the same problem: the second touch-plate button does not work anymore:
 - Program Version	6.4.1.17
-  Build Date & Time	2019-02-14T18:00:57
 - Core/SDK Version	2_4_2/2.2.1(cfd48f3)

The "Toogle 2" button and MQTT commands works normal.

tompipe, 2019-02-15:
I had the same problem a few days back. After some playing about with a few things, I did manage to get it working. I briefly stated above, but didn't give reason...

I can't be sure exactly what did it, but I'm fairly sure it was enabling option51. Setting the module type to generic, and then ensuring the gpio connections match those I found here - https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK,-EU,-and-US

I think even after switching it back to the right module type it still worked. 

It could've been something else, as I don't remember everything I did, but I do recall those above. 

Unfortunately I can't confirm now, as I've re-flashed it with a different fork with the shutter configuration options.

wongnam, 2019-02-16:
> Is it verified that all buttons work with Arduino core 2.3.0 with latest dev version?
> Or is the issue with all core versions?

@Jason2866 I will not go back to 2.3.0 since it make many issue for me in the past. 
- version **6.4.1.17** core 2.4.2 -> has issue with 2nd and 3rd touch button fucntion.
- version **6.4.1.15** core 2.4.2 -> has no issue with any touch button function.
I think you can imagine the issue that it's not related to the core.

arendst, 2019-02-16:
As a workaround use seroption51 1

This re-enables gpio9 and 10 which seem to be disabled by the latest template update. Will fix today.

arendst, 2019-02-16:
SetOption51 1 is not a workaround.

Working on fix...

arendst, 2019-02-16:
Due to the bug the settings for GPIO9 and GPIO10 ar lost. The are being used for button2 (Touch 2CH and Pros) and button3 (Touch 3CH and Pros).

What is a workaround:

- Execute SetOption51 1 (enabling GPIO9 and GPIO10)
- Remember current module and change current module to whatever (Rewriting GPIO9 and GPIO10 values)
- Change to remembered module (Rewriting GPIO9 and GPIO10 values)

Once the fix is released execute SetOption51 0

wongnam, 2019-02-16:
I already revert it back to 6.4.1.15 lastnight, so i will wait for a fix release.

arendst, 2019-02-16:
Thx you for finding the most possible small deviation between (non)-working versions making solving this a lot easier.

andrethomas2, 2019-02-16:
Closing as it was fixed with commit https://github.com/arendst/Sonoff-Tasmota/commit/1fde362998c86c6f9d9bb259afd2252de0cd67e7

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-16 by andrethomas2