arendst/Tasmota #5199: Sonoff basic R2 red led

yonubear, 2019-02-12:
This is more of a question than an issue but i waan sure where else to out it.

On my basic R2s i notice the red led never comes on o ly the green led even when the relay is on So my question is can this be repurposed as a gpio for a second switch?

Thabks,
Yonu

netpok, 2019-02-12:
I could not find a schematic for the basic R2 (or any information on it), but if it is using the same hardware as R1, then it is connected to a dedicated mcu to control that led. You are better off searching an other accessible pin on the ESP, otherwise if it's connected directly to the ESP, then you can use it for almost anything (if it's connected to GPIO0, GPIO2 or TX then it needs to be high at boot time, so no switch only pushbuttons).

andrethomas, 2019-02-12:
@yonubear Please see issue which was previously opened: https://github.com/arendst/Sonoff-Tasmota/issues/4515

closed 2019-02-12 by ascillato2