arendst/Tasmota #5201: Tasmota Device is receiving MQTT Topics from other Devices

helmar74, 2019-02-12:
Hi,

I've got a strange Behaviour
My Wemos Mini is receiving MQTT Topics from other Tasmota Devices!!

Here's the status 0 of this Device:
>12:48:55 CMD: status 0
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS = {"Status":{"Module":18,"FriendlyName":["WemosMini-6"],"Topic":"sonoffWemosMini_6","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"wemosmini6","OtaUrl":"http://192.168.1.2:5555/data/firmwares/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:03:18","StartupUTC":"2019-02-12T11:45:37","Sleep":50,"BootCount":143,"SaveCount":302,"SaveAddress":"F9000"}}
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019.02.07 13:34:06","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":4,"LogHost":"192.168.1.2","LogPort":514,"SSId":["hs.net_EG","sonoffap"],"TelePeriod":3600,"SetOption":["00008009","558180C0","00000000"]}}
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS4 = {"StatusMEM":{"ProgramSize":554,"Free":448,"Heap":5,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","0FFAE79C","00038FA0","23BE1FCE","00003BC0"]}}
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini_6","IPAddress":"192.168.1.166","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"84:F3:EB:0C:2D:65","Webserver":2,"WifiConfig":5}}
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"WemosMini-6","MqttClient":"WemosMini-6","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 12 11:48:55 2019","Local":"Tue Feb 12 12:48:55 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:22","Sunset":"17:24"}}
>12:48:55 MQT: stat/sonoffWemosMini_6/STATUS10 = {"StatusSNS":{"Time":"2019-02-12T12:48:55","SR04":{"Distance":60.105}}}
>12:48:56 MQT: stat/sonoffWemosMini_6/STATUS11 = {"StatusSTS":{"Time":"2019-02-12T12:48:55","Uptime":"0T00:03:18","Vcc":2.713,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"CC:CE:1E:AB:0A:3F","Channel":6,"RSSI":56}}}

And here you can see the MQTT which shouldn't be received from that Device at all:
>Feb 12 12:48:16 sonoffWemosMini_6 ESP-RSL: empfangenes topic tele/sonoff4CH_1/STATE, Datengröße 255, Daten {"Time":"2019-02-12T12:48:15","Uptime":"4T23:06:13","Vcc":3.196,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"CC:CE:1E:AB:0A:3F","Channel":6,"RSSI":66}}
>Feb 12 12:48:23 sonoffWemosMini_6 ESP-SRC: MQTT
>Feb 12 12:48:23 sonoffWemosMini_6 ESP-RSL: empfangenes topic tele/BlitzwoIFSHP2_1/STATE, Datengröße 210, Daten {"Time":"2019-02-12T12:48:23","Uptime":"4T22:33:09","Vcc":3.208,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"hs.net_EG","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":100}}
>Feb 12 12:48:23 sonoffWemosMini_6 ESP-SRC: MQTT
>Feb 12 12:48:23 sonoffWemosMini_6 ESP-RSL: empfangenes topic tele/BlitzwoIFSHP2_1/SENSOR, Datengröße 226, Daten {"Time":"2019-02-12T12:48:23","ENERGY":{"TotalStartTime":"2018-11-04T08:46:00","Total":38.975,"Yesterday":0.055,"Today":0.039,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

Here's the startup message. (Why is the MQTT Topic tele/# used)?
>Mini_6 ESP-MQT: cmnd/sonoffWemosMini_6/POWER =
>Feb 12 12:45:27 sonoffWemosMini_6 ESP-MQT: abonniere cmnd/sonoffWemosMini_6/#
>Feb 12 12:45:27 sonoffWemosMini_6 ESP-MQT: abonniere cmnd/wemosmini6/#
>Feb 12 12:45:27 sonoffWemosMini_6 ESP-MQT: abonniere cmnd/WemosMini-6_fb/#
>Feb 12 12:45:27 sonoffWemosMini_6 ESP-MQT: abonniere tele/#
>Feb 12 12:45:27 sonoffWemosMini_6 ESP-MQT: tele/sonoffWemosMini_6/INFO1 = {"Module":"Generic","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/WemosMini-6_fb/","GroupTopic":"wemosmini6"}

Ist the usage of the Topix tele/# the problem, and how can I disable it?
Thanks a lot!

ascillato, 2019-02-12:
Both devices has the same topic sonoffWemosMini_6?

ascillato, 2019-02-12:
Please, share the status 0 from both devices

helmar74, 2019-02-12:
The Device is receiving the tele/ MQTT Topics from all other devices....

ascillato, 2019-02-12:
That should not happen. There is something wrong in the firmware or in the broker.

Please, try to do a reset 5 on that device or erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins. 

And also delete your mqtt broker database. 

Tasmota never susbscribes to #
That is not in the codebase. So, the issue can be if you had modified the firmware or a retained message in the broker.

helmar74, 2019-02-12:
After reset 5 the error is gone, recovering the settings from a backup brings the error back

ascillato, 2019-02-12:
Great, so, the issue is in the config. Try to add the config mannually

ascillato2, 2019-02-12:
Please, tell us if adding the config mannually solves your issue. Thanks

closed 2019-02-12 by ascillato2

helmar74, 2019-02-12:
That's not possible. After saving the "Device Settings" the Device fails back to defaults!

ascillato, 2019-02-12:
Ok, so there is definitely something wrong with that firmware.

Try to erase all flash with esptool.py as explained in the wiki and flash again with the latest precompiled bins from development branch. 

Setting those gpios should not make your device entering in a bootloop.

arendst, 2019-02-12:
A display binary enabling DisplayMode > 3 will enable receiving all MQTT messages to be able to digest and display the sensor text on the screen.

It indeed subscribes to #

helmar74, 2019-02-12:
But is it correct to subscribe the topic tele/# ?
I mean this will subscribe every "teleperiod topic" from every Device which is connected to the Broker.
It's good, when you want to show values from different sensors, but couldn't it be a problem in a bigger Tasmota Deployment?

arendst, 2019-02-12:
I have over 60 devices active, not all of them have sensors but all five displays never have failed due to message overrun.

helmar74, 2019-02-12:
Ah, great to hear. I've currently about 30 Devices active. ;-)

Just one other question.
As in Core 2.5.0 the webinterface is much faster I decided to upgrade my devices to a version compiled with Core 2.5.0.

I'm running in a FB7590 (Mesh) WLAN environment, with 3 repeaters. (also AVM 1750E)
After the upgrade to the core 2.5.0 Firmware (Tasmota 6.4.1) I do have permanent MQTT reconnects.
Every Time MQTT Reconnects, also the Fritzbox shows, that WIFI connection was lost from this device.
I tried nearly everything (Delete Flash & Flashing by wire, reset 5, new hardware) but nothing really helped.

Everything I tried is documented in #5146 
Do you have any suggestion for me? 
I would like to run Core 2.5.0 in my Fritzbox based MESH environment

Jason2866, 2019-02-13:
Your wifi reconnects are core related as explained in #5146 
NOTHING can be done from Tasmota side to solve. 
Please dont Spam Tasmota Github with this know behaviour from the Arduino core