arendst/Tasmota #5203: iFan02 wrong Speed

sgmelin, 2019-02-12:
Hi,

i figured out, that the Fan Speed is wrong. At Speed 3 (should be fastest) i have a current of 231 Volt, at Speed 1 (should be the slowest), current is 250 Volt. Level 2 is 240 Volt.

andrethomas, 2019-02-12:
> i have a current of 231 Volt

You lost me there...

ascillato, 2019-02-12:
@sgmelin

Sorry, as it is, is correct. The speed 1 is the fastest.

It is needed like that for the motor. It always needs to start into the fastest and then reduce to the others.

ascillato2, 2019-02-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

sgmelin, 2019-02-12:
Problem is, HomeKit uses 33% as Speed 1 and 100% for Speed 3 (using Homebridge and homebridge-mqtt-fan-tasmota plugin).

ascillato2, 2019-02-22:
Ok.

@arendst, any comment?

arendst, 2019-02-22:
Home kit???

ascillato, 2019-02-22:
Yes, but both (@netRunner0 and @sgmelin) said that in iFan the speeds should be
1 - min speed
3 - max speed

netRunner0, 2019-02-22:
Yes, that’s how it was in the native application)

And the homebridge plugin also works incorrectly (vice versa)

arendst, 2019-02-22:
What's the issue here? Tasmota iFan support hasn't changed so didn't speed. It has no relation with iTead software so it is what it is for many happy users probably not using homekit.

netRunner0, 2019-02-22:
I connected the sonof to 120v and now it really works like nuzhe. From the quietest mode to maximum (noisy) speed

andrethomas, 2019-02-22:
@netRunner0 I'm not sure if I understand your response, but I guess it means that you got it working as expected.

ascillato2, 2019-02-22:
Closing this issue as the misunderstanding has been solved.

closed 2019-02-22 by ascillato2

sgmelin, 2019-02-22:
I used the Sonoff Web Page to measure the voltage per Speed level to avoid HomeKit plugin issue. Again, according to Sonoff, 1 is the slowest and 3 the fastest mode. Closing the thread after an hour without changing the code does not solve anything... please change the code. It is useless at the moment...

andrethomas, 2019-02-22:
> please change the code.

Not likely to be changed since it is the way as it is now since it was developed.

> It is useless at the moment

You don't have to agree with anyone here... nobody can force you to be right :)

sgmelin, 2019-02-23:
the fact, nobody did the effort to measure the voltage per speed level seems to be the root cause here... to suppose you’re right, just because you did it since the beginning is no proof to be right. The measurement of the power output is a proof, actually. If anybody does the same and convince me I’m wrong, I do no longer insist a code change. So, buy a voltage meter, connect a fan and do a measurement by your own. Than look into the original Sonoff docs and compare with your measures...

netRunner0, 2019-02-24:
This problem occurs.

Normally, the iphone only works on 120 volts. Only at this voltage, I feel the difference in the work of the hood, while when powered from 220 volts I did not see the difference changing the modes.

https://github.com/rgavril/homebridge-mqtt-fan-tasmota
Here is a homekit plugin. It is with him that the following problem is observed, when at the minimum speed in the homekit the fan runs at maximum and vice versa.

I looked at the plug-in and realized that it doesn’t calculate anything, but simply takes values from the mqtt.

sgmelin, 2019-02-24:
Hi,

As I mentioned before, I did the measurement directly by steering via the Tasmota Web Interface. Homekit only switches the modes….