arendst/Tasmota #5204: Wifi disconnects / mqtt disconnets / webserver stops working help needed.

savage872, 2019-02-12:
_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : Sonoff T1 UK / Sonoff CH4 PRO
- [x] _Development/Compiler/Upload tools used_ :OTA
- [x ] _If a pre-compiled release or development binary was used, which one?_ :6.4.1
- [x ] _You have tried latest release or development binaries?_ :6.4.1.14
- [ x] _Provide the output of command_``status 0`` :
```
16:47:10 MQT: stat/B2/STATUS = {"Status":{"Module":28,"FriendlyName":["Daniels Bedroom"],"Topic":"B2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
16:47:10 MQT: stat/B2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"Power on","Uptime":"0T03:53:20","StartupUTC":"2019-02-12T12:53:50","Sleep":50,"BootCount":53,"SaveCount":318,"SaveAddress":"FB000"}}
16:47:10 MQT: stat/B2/STATUS2 = {"StatusFWR":{"Version":"6.4.1.14(b769158-sonoff)","BuildDateTime":"2019-02-07T10:41:04","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
16:47:10 MQT: stat/B2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home Sweet Home",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A029","280500000100000000000000000000000000","00000001"]}}
16:47:10 MQT: stat/B2/STATUS4 = {"StatusMEM":{"ProgramSize":553,"Free":448,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
16:47:10 MQT: stat/B2/STATUS5 = {"StatusNET":{"Hostname":"B2","IPAddress":"192.168.0.112","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:99:95:1D","Webserver":2,"WifiConfig":3}}
16:47:10 MQT: stat/B2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"B2","MqttClient":"B2","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:47:10 MQT: stat/B2/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 12 16:47:10 2019","Local":"Tue Feb 12 16:47:10 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+00:00","Sunrise":"07:03","Sunset":"17:05"}}
16:47:10 MQT: stat/B2/STATUS10 = {"StatusSNS":{"Time":"2019-02-12T16:47:10"}}
16:47:10 MQT: stat/B2/STATUS11 = {"StatusSTS":{"Time":"2019-02-12T16:47:10","Uptime":"0T03:53:20","Vcc":3.504,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Home Sweet Home","BSSId":"E4:F4:C6:79:C6:C8","Channel":11,"RSSI":96}}}
```

16:42:14 WIF: Checking connection...
16:42:14 WIF: Connected
16:42:34 WIF: Checking connection...
16:42:34 WIF: Connected
16:42:40 MQT: Attempting connection...
16:42:40 MQT: Connect failed to 192.168.0.100:1883, rc -2. Retry in 10 sec
16:42:54 MQT: Attempting connection...
16:42:54 MQT: Connected
16:42:54 MQT: tele/B2/LWT = Online (retained)
16:42:54 MQT: cmnd/B2/POWER = 
16:42:54 MQT: Subscribe to cmnd/B2/#
16:42:54 MQT: Subscribe to cmnd/sonoffs/#
16:42:54 MQT: Subscribe to cmnd/B2_fb/#

I'm using several devices in my house using tasmota. the sonoff basics/rf bridge/wemos d1 mini are ok
but I can't figure out why the wifi keeps dropping off and mqtt disconnects as a result of it
I've monitored for some time, used weblog 4 and changed firmware version but I can't seem to figure this one out. I do not have a problem with wifi router. When the devices disconnect, they do not disconnect at the same time, just random.
I also have a big problem with my 4x Sonoffs T1 UK which I can't access webserver after a couple of hours after a reboot. It was quite annoying having lift the face plates off and pressing reset each time. The webservers become unresponsive in less than a day. 
I did a bit o research and learned that you can send a mqtt message and reboot device via mqtt, so know that's what I'm doing did it's still not as it should be.
So, to summarise, The touch button works, mqtt works but no web access and it also doesn't respond to ping.

I hope someone has an answer for my 2 issues as this is becoming very unpopular with the wife :)

ascillato, 2019-02-12:
Hi,

This is a duplicated issue. 

Please read the following link of the wiki for the source of the problem and the solution. 

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

If you need further assistant please find us in the Tasmota Support Chat. 

Thanks

ascillato2, 2019-02-12:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-12 by ascillato2

savage872, 2019-02-13:
I can confirm that choising core 2.3.0 makes it a lot more stable... thanks for you help!