arendst/Tasmota #5206: Add CFG_HOLDER to one of the status messages

Ysbrand, 2019-02-12:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**

I have no means of determine the current CFG_HOLDER that is used for the permanent settings

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Add the current CFG_HOLDER  value to either "status 1" or "status 4"

ascillato2, 2019-02-12:
Hi,

Sorry, this is a duplicated issue. 

The request was dismissed so as to not send a constant every teleperiod.

closed 2019-02-12 by ascillato2

Ysbrand, 2019-02-13:
I don't understand this. I'm not asking for a 'tele' message, but for a stat message.

arendst, 2019-02-13:
I can add it easily. Will do soon.

Ysbrand, 2019-02-13:
Thanks Theo, much appreciated!