arendst/Tasmota #5208: Strange console messages from eMylo Smart Switch TYWE2S

DillonJ, 2019-02-12:
Hi folks,

I have flashed two "eMylo Smart switches" with Tasmota using Sonoff.bin. I believe they are using Tuya TYWE2S. I am using the default configuration for now.  Strangely enough, in the default configuration, the toggle button actually toggles the relay of the eMylo smart switch. Is it just coincidence that the relay is controlled by the same GPIO pin as a Sonoff? Or is there something more sophisticated happening?

Is there a better way to figure out what is connected to what other than trial and error.... I did try configuring as Generic (18) and managed to turn the LED on and off and figure out which GPIO the button was attached to... but then I started seeing strange commands in the console. I reverted to the default config but am seeing the same. I flashed a second device and am seeing similar console messages without changing the config:

```
22:06:53 MQT: stat/sonoff-4534/RESULT = {"Command":"Unknown"}
22:07:05 CMD: .?
22:07:05 MQT: stat/sonoff-4534/RESULT = {"Command":"Unknown"}
22:07:31 CMD: 
22:07:31 MQT: stat/sonoff-4534/RESULT = {"Command":"Unknown"}
22:07:33 CMD: 
22:07:33 MQT: stat/sonoff-4534/RESULT = {"Command":"Unknown"}
22:07:40 CMD: 
22:07:40 MQT: stat/sonoff-4534/RESULT = {"Command":"Unknown"}
22:07:53 CMD: 
22:07:53 MQT: stat/sonoff-4534/RESULT = {"Command":"Unknown"}
22:09:40 MQT: tele/sonoff-4534/STATE = {"Time":"2019-02-12T22:09:40","Uptime":"0T02:45:16","Vcc":3.334,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Office-Wifi","BSSId":"0E:41:58:00:CD:25","Channel":10,"RSSI":88}}
22:10:31 CMD: 
22:10:31 MQT: stat/sonoff-4534/RESULT = {"Command":"Unknown"}
22:12:06 CMD: 
22:12:06 MQT: stat/sonoff-4534/RESULT = {"Command":"Unknown"}
```

DillonJ, 2019-02-12:
Seems also to be losing wifi connection and reconnecting:

```
22:44:40 DNS: Initialized
22:44:40 MQT: Attempting connection...
22:44:40 MQT: Connected
22:44:40 MQT: tele/sonoff-4013/LWT = Online (retained)
22:44:40 MQT: cmnd/sonoff-4013/POWER = 
22:48:06 MQT: Attempting connection...
22:48:06 MQT: Connect failed to 192.168.1.22:1883, rc -2. Retry in 10 sec
22:48:08 WIF: Connecting to AP1 Livebox-08A0 in mode 11N as sonoff-4013-4013...
22:48:12 WIF: Connected
22:48:12 DNS: Initialized
22:48:22 MQT: Attempting connection...
22:48:23 MQT: Connected
22:48:23 MQT: tele/sonoff-4013/LWT = Online (retained)
22:48:23 MQT: cmnd/sonoff-4013/POWER = 
22:48:58 MQT: tele/sonoff-4013/STATE = {"Time":"2019-02-12T22:48:58","Uptime":"0T03:15:21","Vcc":3.475,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Livebox-08A0","BSSId":"90:4D:4A:B1:08:A0","Channel":6,"RSSI":80}}
22:51:24 DNS: Initialized
22:51:24 MQT: Attempting connection...
22:51:25 MQT: Connected
22:51:25 MQT: tele/sonoff-4013/LWT = Online (retained)
22:51:25 MQT: cmnd/sonoff-4013/POWER = 
22:53:58 MQT: tele/sonoff-4013/STATE = {"Time":"2019-02-12T22:53:58","Uptime":"0T03:20:21","Vcc":3.474,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Livebox-08A0","BSSId":"90:4D:4A:B1:08:A0","Channel":6,"RSSI":80}}
```

ascillato2, 2019-02-12:
Hi,

Please, for support use the Tasmota Support Chat. Github issues are for bug reporting. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-12 by ascillato2

Jason2866, 2019-02-13:
Try `reset 5`. This erases flash (not Tasmota) and reset to defaults. Only wifi credentials are not erased.
Erasing flash deletes wifi calibration. 
If this is done devices does wifi calibration again and store actual settings in flash.
So erasing flash is always a good idea, if strange errors occur...