arendst/Tasmota #5211: H801 set color blinks once

mtmayr, 2019-02-13:
My H801 (20) works quite fine with firmware "Sonoff-Tasmota 6.4.1.8". Except for some reboots once a while. But when I upgraded to "6.4.1.15", setting a color results in showing the correct color, after 1s shortly showing another color (sometimes red, sometimes black, sometimes blue) in between for 2/10s, before showing the correct color again. It seems to be a part of the switching process. After that, the color is steady on an correct. No separate MQTT-message is sent. It seems to be dependant on the color, as some colors like #ff0000, #ffff00, #0000ff work fine, but e.g. #800050 does not. No complicated setup, just entering the color on console or via MQTT results in that behaviour!

Do you have any suggestions for me?

- [x] Searched the problem in issues and in the wiki : yes
- [x] Hardware used : H801 WiFi from AliExpress with MQTT
- [x] Development/Compiler/Upload tools used : Arduino 1.8.7 with Teensyduino 1.45
- [x] If a pre-compiled release or development binary was used, which one? : self compiled
- [x] You have tried latest release or development binaries? : no, src version: 6.4.1.15
- [x] Provide the output of command ``status 0`` :
```
11:03:19 MQT: stat/sonoff_33BC6868/STATUS = {"Status":{"Module":20,"FriendlyName":["Sonoff"],"Topic":"sonoff_33BC6868","ButtonTopic":"","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:03:58 MQT: tele/sonoff_33BC6868/STATE = {"Time":"2019-02-13T11:03:58","Uptime":"0T00:05:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Dimmer":50,"Color":"7F004F0000","HSBColor":"323,100,50","Channel":[49,0,30,0,0],"CT":499,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":x,"RSSI":88}}
```

**To Reproduce**
git clone
my_user_config.h: #define USE_CONFIG_OVERRIDE
my_user_config.h: #define FIRMWARE_CLASSIC
user_config_override.h: #define MODULE                 H801
user_config_override.h: increase CFG_HOLDER, 
flash H801 with soldered pins with Arduino IDE and jumper J3
disconnect+remove J3 and connect to external power
connect to web-interface/login
console command: "color #ff0000"
no blinking result, just red
console command: "color #800050"
fail result: displaying purple 1s, then black 2/10s, then stays purple

**Expected behavior**
just display #800050 without changing color
(it seems to work correctly with colors like #ff0000, #ffff00, #0000ff)

**Screenshots**
none, should I upload a video?

**Additional context**
Worked in 6.4.1.8, stopped working afterwards

Jason2866, 2019-02-13:
Please erase flash via esptool.  Try latest precompiled binary from http://thehackbox.org/tasmota/
No OTA upload. Bad settings can be left in flash!
If error is still there provide `Status 0` output.

mtmayr, 2019-02-13:
Thanks for reply Jason2866!
I tried to esptool erase_flash but that lead to an error: "ESP8266 ROM does not support function erase_flash."
When programming to thehackbox-version [1] you lead me, the color switching works fine!! ("esptool write_flash" erased the flash too I guess)
When programming again to my compiled version [2], it has the same unwanted behaviour again! Interesting!

[1] 6.4.1.16(b9f6bf4-sensors)-2_4_2
[2] 6.4.1.16(sensors)-2_5_0_BETA2
PS: I added the "status 0" above

mtmayr, 2019-02-13:
Upgrading Board Manager to "2.5.0-beta3" fixed it. Thank you!
(Arduino IDE: Tools / Board / Boards Manager / esp8266 "ESP8266 Community")

ascillato2, 2019-02-13:
Great to know that it is working now for you.

Closing this issue as it is solved. Thanks

closed 2019-02-13 by ascillato2