arendst/Tasmota #5212: Option for DisplayMode 2 to hidden Topic name for local Sensor value.

wongnam, 2019-02-13:
May i know if any Option for DisplayMode 2 to hidden topic name for local sensor value ?
it's no need to have topic name for local sensor information to save display space to show the important value.

[image]

arendst, 2019-02-13:
There is none but this is a good idea.

Will follow up (someday).

andrethomas2, 2019-02-26:
Closing this issue as it the necessary changes made for this request has been merged into development.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-26 by andrethomas2