arendst/Tasmota #5213: MQ-135 possible with tasmota and d1 mini?

emersonicus, 2019-02-13:
Hi,

Does tasmota support MQ-135 gas sensor? If yes how can I enable it?

thanks!

ascillato, 2019-02-13:
Hi,

That sensor has an analog output. So, you can attach it directly to tha A0 pin and you need to flash the Tasmota firmware called sonoff-sensor.bin to use the adc. 

Then the measurements can be sent to your home automation software. Or you can make rules to do whatever you want. 

If you need further assistant please find us in the Tasmota Support Chat. Thanks

ascillato2, 2019-02-13:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-13 by ascillato2

joba-1, 2019-02-13:
Hi Adrian,

I tend to disagree with your reasoning here. It is a bit like saying „We
have i2c support. Hook up your oled display, add some rules an you‘re good“.

Getting reasonable values out of those sensors is not that easy. It needs
calibration and a bit of math. Not sure rules could do all that.
I have these sensors and I’m not there yet...
-- 
iphone so i typo

ascillato, 2019-02-13:
@joba-1

Hi,

I agree with you that with a driver implemented in Tasmota, all that config will be a lot easier. Until someone makes that, it can be used the direct readings from the A0 pin and do the math in your home automation software.
Another available option is to do the math (explained in the datasheet of MQ-135) using rules. The math required and the conditionals can be done by rules. With the open PR for expression evaluations will be easier.

Anyway, the Tasmota project is open for PRs if you want to make it easier.

In my comment I said:

>If you need further assistant please find us in the Tasmota Support Chat. Thanks

And I meant it. If you need help on doing that, we can help you. Usually I just take some more time and write a rules' example for doing that. Today I'm not at home to do and test that. So, that I let to you to start trying to make the implementation in your home automation software or making rules.

Anyway, if you want to make a driver, we can also assist you on that.