arendst/Tasmota #5214: Unable to connect to WebServer on 10.1.x.x/16 network

gazoodle, 2019-02-13:
**Describe the bug**
If your DCHP server issues addresses in 10.1.x.x/16 range then the Tasmota WebServer is **not** accessible after the device connects to the WLAN and you cannot access the configuration address http://10.1.x.x/cn

If the DCHP sever issues addresses in 192.168.x.x/16 range then the Tasmota WebServer *IS* accessible after the device connects to the WLAN and you can navigate to http://192.168.x.x/cn

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff Basic
- [X] _Development/Compiler/Upload tools used_ : None
- [X] _If a pre-compiled release or development binary was used, which one?_ : sonoff.bin
- [X] _You have tried latest release or development binaries?_ : No
- [X] _Provide the output of command_``status 0`` :
```
status 0
14:09:50 CMD: status 0
14:09:50 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:09:50 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:18","StartupUTC":"2019-02-13T13:09:32","Sleep":50,"BootCount":34,"SaveCount":107,"SaveAddress":"F8000"}}
14:09:50 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:09:50 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TENSIXTEEN",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000001"]}}
14:09:50 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
14:09:50 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1207","IPAddress":"10.1.24.2","Gateway":"10.1.0.1","Subnetmask":"255.255.0.0","DNSServer":"10.1.0.194","Mac":"DC:4F:22:95:44:B7","Webserver":2,"WifiConfig":2}}
14:09:50 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9544B7","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:09:50 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 13 13:09:50 2019","Local":"Wed Feb 13 14:09:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:02","Sunset":"18:06"}}
14:09:50 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-13T14:09:50"}}
14:09:50 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-13T14:09:50","Uptime":"0T00:00:18","Vcc":3.255,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"TENSIXTEEN","BSSId":"<redacted>","Channel":1,"RSSI":92}}}
14:09:54 MQT: Attempting connection...
```

**To Reproduce**
Set your Wifi DHCP server to issue addresses in 10.1.0.100 -> 10.1.255.254 range.
Configure Tasmota to use this Wifi
Restart device.
Wait until you get the prompt on the serial port of the IP address the WebServer is listening on
Attempts to PING or connect to the device will fail.

**Expected behavior**
I have a dream. A dream where all my four little IP address nibbles live in a network where they will not be judged by their value ... :-)

**Screenshots**
I think this is pretty concise, but I can provide more data if needed.

**Additional context**

**(Please, remember to close the issue when the problem has been addressed)**

digiblur, 2019-02-14:
Mine live in a 10.x range of IPs just fine, now I don't recall what size it is though.

gazoodle, 2019-02-14:
I can say that the MQTT client and broker connection always work fine (perhaps because the connection is outbound from the Sonoff), it's just that you can't access inbound to the device. I'll have more of a play around and see.

I can add that I have tried 10.1.0.x/24 and that doesn't work either. I don't think I can try 10.x.x.x/8 (which is what IANA reserved the range for) because I think my DHCP server gets upset about the number of leases it would have to deal with!

I'll try a 172.16.x.x/16 range to see what happens.

gazoodle, 2019-02-14:
Hmmm. Very strange. It all works fine on 172.16.x.x/16 AND it also works on a 10.0.0.x/24 network too. It's just my 10.1.x.x/16 (or 10.1.0.x/24) that doesn't work.

arendst, 2019-02-14:
From your last answer I suggest to try another core version. You currently use 2.4.2 but might want to try 2.3.0 or 2.5.0. low level TCP/IP handling (as is DHCP) is done by the core software not by Tasmota.

gazoodle, 2019-02-14:
So to be clear the MQTT client/broker connection always works. This is about connecting to the Webserver.

```
10.0.0.250/24 - WORKS
10.1.0.248/16 - FAILS
172.16.0.230/16 - WORKS
192.168.1.247/16 - WORKS
```

gazoodle, 2019-02-14:
I just did one more test 
````
10.20.0.250/24 - WORKS
```

I'm not convinced now that this is a Tasmota issue given that evidence but what the problem it is on my network I don't know ... sigh

gazoodle, 2019-02-14:
I'm going to close this issue because I don't think it's anything to do with Tasmota or the device. Admittedly it's strange because I have other Webservers in the 10.1.x.x/16 network that are fine, but if I have the device on 10.1.0.203/16 and I have another network on 10.2.x.x/16 and I bridge between the two then a browser on the 10.2.x.x network can connect fine, it's just browsers in the 10.1.x.x network that can't and that's weird but I don't see how this can be anything to do with the device itself. I've tried some tcpdumps from my AP but there are no packets even being delivered when on the same 10.1.x.x subnet. I will post a final reply when I've worked out what the heck is going on in case this helps someone else out.

closed 2019-02-14 by gazoodle

gazoodle, 2019-02-14:
Well, that was an episode in PEBKAC if ever I saw one. I have a Ubiquiti Unifi and had previously enabled the **Block LAN to WLAN Multicast and Broadcast Data** feature. As soon as I disabled this (to check) the device was accessible. I'm not sure I fully understand why a multicast/broadcast packet is required for a point-to-point connection, and other webservers in the same sub-net work fine. This has actually been a good process overall because it has forced me to actually get around to creating a separate sub-net for my home automation devices which is a good thing anyway