arendst/Tasmota #5215: NX-SP201

awk2019, 2019-02-13:
I have flashed two NX-SP201 dual switch with Tasmota via tuya-convert on one and cracked on open for wired serial flash.
When I use power monitoring via a custom template(HLW8012) I get a software watchdog resets about every 40-60 minutes. Do any one have any ideas on this. 

Pin out per tuya-convert compatible device list.
GPIO0: Led1i, GPIO4: Button1, GPIO5: HLW8012 CF, GPIO12: Relay1, GPIO13: Button2, GPIO14: HLWBL CF1, GPIO15: Relay2, GPIO16 HLWBL SEL

Jason2866, 2019-02-13:
`Status 0` is missing. I have no crystal ball :-)

awk2019, 2019-02-13:
What information is needed from status 0

Jason2866, 2019-02-13:
All. Just copy paste from webconsole.

awk2019, 2019-02-13:
18:26:53 CMD: status 0
18:26:53 MQT: stat/DualOutLet2/STATUS = {"Status":{"Module":61,"FriendlyName":["DualOutlet2A","DualOutlet2B"],"Topic":"DualOutLet2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:26:53 MQT: stat/DualOutLet2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:17:30","StartupUTC":"2019-02-13T17:09:23","Sleep":50,"BootCount":81,"SaveCount":454,"SaveAddress":"F7000"}}
18:26:53 MQT: stat/DualOutLet2/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-02-13T11:03:23","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
18:26:53 MQT: stat/DualOutLet2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SECRET",""],"TelePeriod":15,"SetOption":["00208009","558180C0","00000400"]}}
18:26:53 MQT: stat/DualOutLet2/STATUS4 = {"StatusMEM":{"ProgramSize":571,"Free":432,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","1FDAE794","000383A0","23B617CE","0001BBC0"]}}
18:26:53 MQT: stat/DualOutLet2/STATUS5 = {"StatusNET":{"Hostname":"Benlink","IPAddress":"192.168.1.90","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"SECRET","Webserver":2,"WifiConfig":4}}
18:26:53 MQT: stat/DualOutLet2/STATUS6 = {"StatusMQT":{"MqttHost":"SECRET","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F9D903","MqttUser":"SECRET","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:26:53 MQT: stat/DualOutLet2/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 13 17:26:53 2019","Local":"Wed Feb 13 18:26:53 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:02","Sunset":"18:06"}}
18:26:53 MQT: stat/DualOutLet2/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:26:53 MQT: stat/DualOutLet2/STATUS10 = {"StatusSNS":{"Time":"2019-02-13T18:26:53","ENERGY":{"TotalStartTime":"2019-02-07T03:09:41","Total":0.166,"Yesterday":0.040,"Today":0.125,"Power":51,"ApparentPower":63,"ReactivePower":37,"Factor":0.81,"Voltage":123,"Current":0.514}}}
18:26:53 MQT: stat/DualOutLet2/STATUS11 = {"StatusSTS":{"Time":"2019-02-13T18:26:53","Uptime":"0T00:17:30","Vcc":3.423,"SleepMode":"Normal","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SECRET","BSSId":"SECRET","Channel":4,"RSSI":100}}}

Jason2866, 2019-02-13:
Try precompiled version from http://thehackbox.org/tasmota/sonoff.bin
Probably your selfcompiled is buggy...
Before flashing Tasmota erase whole flash with esptool erase_flash

awk2019, 2019-02-13:
I had the same problem with the precompiled dev tasmota on both devices,and the released version dose not have a template for the nx-sp201 dual plug switch. Also its a glued to together device.

Jason2866, 2019-02-13:
Try reset 2  Probably fragments left in flash from original firmware causes your issues
It erases all flash behind tasmota and resets to defaults.

awk2019, 2019-02-13:
After tasmota update and reset 2, I configured the device per 
GPIO0: Led1i, GPIO4: Button1, GPIO5: HLW8012 CF, GPIO12: Relay1, GPIO13: Button2, GPIO14: HLWBL CF1, GPIO15: Relay2, GPIO16 HLWBL SEL
then when I celebrated the voltage it started resting every 2-10 seconds.
When I use it with out power monitoring it will work with out a problem.

Jason2866, 2019-02-13:
"SP201",
     GPIO_LED1_INV,    // GPIO00 Led (1 = On, 0 = Off) - Power status
     GPIO_NRG_SEL_INV, // GPIO03 HLWBL HJL-01 Sel output (1 = Voltage)
     GPIO_KEY1,        // GPIO04 Button 1
     GPIO_HJL_CF,      // GPIO05 HLW8012 BL0937 or HJL-01 CF power
     GPIO_REL1,        // GPIO12 Relay1 (0 = Off, 1 = On)
     GPIO_KEY2,        // GPIO13 Button 2
     GPIO_NRG_CF1,     // GPIO14  HLWBL BL0937 or HJL-01 CF1 current / voltage
     GPIO_REL2,        // GPIO15 Relay2 (0 = Off, 1 = On)

Jason2866, 2019-02-13:
I dont see you have configured GPIO3

awk2019, 2019-02-14:
no good I tried multiple variations of the sensor pin settings and reset 2 the NX-SP201 with no improvement, I have two devices behaving exactly the same way.
I can get power readings for a 30 to 60 minutes before a software watchdog reset, and the power readings are correct in comparison to a KILL A WATT meter with BL0937 sensor.

Jason2866, 2019-02-14:
I am out of ideas. Since we (support team) dont own this device we cant help you further...

ascillato2, 2019-02-15:
Please, try http://thehackbox.org/tasmota/020500/sonoff.bin and then do a reset 5

awk2019, 2019-02-15:
no good ether, I just going to give up on power monitoring for now.

Jason2866, 2019-02-15:
Have you tried the preconfigured template for this model?
It is in latest dev firmware....

DavinKD, 2019-02-16:
I just flashed a couple of these and have the same issue when power monitoring is configured.

Jason2866, 2019-02-16:
Out of luck, since we dont have this device..
Maybe someone finds a solution.

DavinKD, 2019-02-16:
I'm digging into the code and will report back if I figure out the issue.

DavinKD, 2019-02-16:
Started up Syslog server and caught this.

sonoff-7484 ESP-RSL: INFO3 = {"RestartReason":"Fatal exception:2 flag:2 (EXCEPTION) epc1:0x3ffff140 epc2:0x00000000 epc3:0x00000000 excvaddr:0x3ffff140 depc:0x00000000"}

DavinKD, 2019-02-16:
Not 100% sure yet, but the issue seems to be caused by running a status 0 command, perhaps while it's in the middle of taking a power reading.  When I removed that command from my SmartThings device handler, the issue seems to have stopped.  I need to let it run for a while to be sure.

DavinKD, 2019-02-17:
Ok, so when I disable the 200ms ticker for energy checking, I can run status 0 many times with no crashing.  There's definitely something there.

DavinKD, 2019-02-18:
FYI, today I rebuilt from the latest dev branch and it all seems to be working.  
/shrug

Using the following GPIO Pin arrangement (please note I prefer to control the blue LED as a separate relay instead of it being GPIO_LED1_INV since there is already a red led that turns on with the relays).

  { "SP201",           // Orange Tech 2 in 1 Smart Socket Wifi Mini Outlet  with Energy Monitoring
                       // https://www.amazon.com/gp/product/B07L63S731
                       // used https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-OTA for flashing Tasmota via OTA
     GPIO_REL3_INV,    // GPIO00 Led (1 = On, 0 = Off) - Power status
     0, 0,
     GPIO_NRG_SEL_INV, // GPIO03 HLWBL HJL-01 Sel output (1 = Voltage)
     GPIO_KEY1,        // GPIO04 Button 1
     GPIO_HJL_CF,      // GPIO05 HLW8012 BL0937 or HJL-01 CF power
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_REL1,        // GPIO12 Relay1 (0 = Off, 1 = On)
     GPIO_KEY2,        // GPIO13 Button 2
     GPIO_NRG_CF1,     // GPIO14  HLWBL BL0937 or HJL-01 CF1 current / voltage
     GPIO_REL2,        // GPIO15 Relay2 (0 = Off, 1 = On)
     0, 0
  }

Jason2866, 2019-02-18:
Could you add your setup to the wiki with a template and close this issue?
Thank you
https://github.com/arendst/Sonoff-Tasmota/wiki/User-created-templates

DavinKD, 2019-02-18:
Sorry, I didn't open the issue.  I'll get the template added shortly unless @balvant813 already did.  I copied from his merge request.

Jason2866, 2019-02-18:
Ahh, yes. we will close. Thx for adding in wiki

andrethomas2, 2019-02-18:
Closing this issue as requested above.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-18 by andrethomas2

DavinKD, 2019-02-18:
No worries.  Wiki updated.  Hopefully I did it correctly.  There's many devices on Amazon which are the same, but I just linked one of them.

DavinKD, 2019-02-18:
@Jason2866

Should the template functionality work the same as modifying the sonoff_template.h file as in the change by @balvant813?

https://github.com/arendst/Sonoff-Tasmota/pull/5180/commits/dae9ecb4d66d8eea897742d0f488f5ce303aa13c

Reason I ask, is because when I compile my own bin file from the latest dev branch using this template file, everything works great.  However, when I went to test the template I created in the wiki, the watchdog reboots re-appeared.

Jason2866, 2019-02-18:
Should be the same... I will investigate tomorrow

DavinKD, 2019-02-18:
@Jason2866 

I was wrong about which branch @balvant813 was working from.  His is actually 6.4.1.15.  Maybe something with the HLW8012/HLJ-01 broke between those versions?  I can consistently compile his check-in and it works fine with my device.  The latest (.17) gives the watchdog error whether I use the template, set the pins manually for a generic device or change the sonoff_template.h similar to his release.

All this being said, I have a version that works now, and maybe it would be a waste of time for you to investigate, but I'm happy to pay for you to get one to test with.  They're only $16.

arendst, 2019-02-19:
@DavinKD few questions:

- on what core version do you get these watchdog errors?
- when you received exceptions did they always have the same address (as long as you used the same binary)?

DavinKD, 2019-02-19:
@arendst 

If I'm reading things correctly in platformio.ini, I believe they are both using 2.4.2

[core_active]
; Select one core set for platform and build_flags
;platform                  = ${core_2_3_0.platform}
;build_flags               = ${core_2_3_0.build_flags}
platform                  = ${core_2_4_2.platform}
build_flags               = ${core_2_4_2.build_flags}
;platform                  = ${core_2_5_0.platform}
;build_flags               = ${core_2_5_0.build_flags}
;platform                  = ${core_stage.platform}
;build_flags               = ${core_stage.build_flags}

I very rarely get anything more than "Software Watchdog" or "Hardware Watchdog".  Below are a few exceptions.

2/16/2019 4:59 PM,Debug,192.168.1.148,sonoff-7484 ESP-RSL: INFO3 = {""RestartReason"":""Fatal exception:0 flag:2 (EXCEPTION) epc1:0xc003b015 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000""}

2/16/2019 2:51 PM,Debug,192.168.1.148,sonoff-7484 ESP-RSL: INFO3 = {""RestartReason"":""Fatal exception:0 flag:2 (EXCEPTION) epc1:0xc003201d epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000""}

2/16/2019 1:56 PM,Debug,192.168.1.148,sonoff-7484 ESP-RSL: INFO3 = {""RestartReason"":""Fatal exception:0 flag:2 (EXCEPTION) epc1:0x313a2267 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000""}

2/16/2019 1:18 PM,Debug,192.168.1.148,sonoff-7484 ESP-RSL: INFO3 = {""RestartReason"":""Fatal exception:0 flag:2 (EXCEPTION) epc1:0x3a22656d epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000""}

2/16/2019 12:57 PM,Debug,192.168.1.148,sonoff-7484 ESP-RSL: INFO3 = {""RestartReason"":""Fatal exception:0 flag:2 (EXCEPTION) epc1:0x3a22656d epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000""}

arendst, 2019-02-19:
All exceptions are 0 - Illegal Instruction. This normally looks like a power supply failure.

The watchdogs are harder to trace...

To be certain, these watchdogs do appear out of the blue or is the device acting on commands like ``status 0``?

DavinKD, 2019-02-19:
It only seems to occur when running a status 0 and only while energy monitoring is enabled (GPIO set for the HLW8012).  You have to run a few status 0 commands, but it will consistently crash after doing it 3-10 times.  I usually wait about 3-5 seconds in between.  However, I have seen it happen when I'm not testing and my SmartThings device handler calls the status 0 (every 5 minutes).  I had a lamp hooked up and let it run for hours and I occasionally would hear the relay click and the light flash off then on.  That's what clued me in that something was wrong.  I can of course, speed up the failure by sending a bunch of status 0 command in close proximity.

arendst, 2019-02-19:
Ok. Makes sense. I experience watchdogs too on a simple wemos device with status 0 but only on core 2.5.0.

So I don't think it's an issue with your device but more a problem in the code either tasmota, core or sdk.

I'll have to dig deep.

DavinKD, 2019-02-19:
Thanks.  I've been unable to make it crash on .15 (from @balvant813) with core 2.4.  I've not checked extensively as to whether he made any other changes besides to the template.

arendst, 2019-02-19:
Just tried TasmoAdmin (which sends status 0 commands as does SmartThings) and was greeted with a lot of clicking relays of devices I updated yesterday to 6.4.1.17 and core 2.4.2 as results of watchdogs...

I'll have to dig deep fast...

DavinKD, 2019-02-19:
Wish I knew the code well enough to be of use.  I'm still figuring out where everything is and what it all does.

Jason2866, 2019-02-19:
@arendst  may you take a look here https://github.com/esp8266/Arduino/commit/7745e99046bcbfbee5058376f30ca213146c3439

arendst, 2019-02-19:
@Jason2866 I know what's going on there but for now I noticed we have an issue with both core 2.4.2 and 2.5.0 related to tasmota versions between 6.4.1.14 (ok) and 6.4.1.16 (faulty).

@andrethomas  and @ascillato2 know about it.

andrethomas, 2019-02-19:
Its something which causes a software watchdog reset from the core in core_esp8266_main.cpp:134 loop_task

That's all we know for now... to be continued...

jziolkowski, 2019-02-20:
I've flashed my 4CH with 6.4.17 on 2.5.0 and it restarted on statuses 3,4,5 and 9. Other were fine.

KPK75, 2019-02-20:
No watchodgs when command "status 0" is sent via MQTT ( 6.4.1.17 / 2.5.0 ).

arendst, 2019-02-21:
Thx for your input. What we know sofar is that it is initiated by HTTP only; Serial and MQTT are fine.

Investigating.

jziolkowski, 2019-02-21:
The statuses I wrote above were initiated via MQTT. I'll re-flash the device with core 2.5.0 and test again.

arendst, 2019-02-21:
@jziolkowski I now know the source of the "hangs" is not http initiated.

It fails while executing the status mqtt result message way before sending the results to http console or client.

Investigating.

arendst, 2019-02-21:
Got something with debug enabled and memfree set to 1:

```
11:50:17 DBG: FreeRam 22392, FreeStack 1040 (MqttPublishDirect)
11:50:17 MQT: stat/wemos2/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(sonoff)","BuildDateTime":"2019-02-21T11:46:57","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
```
and
```
11:59:47 DBG: FreeRam 23088, FreeStack 2224 (MqttPublishDirect)
11:59:47 MQT: stat/wemos2/STATUS2 = {"StatusFWR":{"Version":"6.4.1.14(sonoff)","BuildDateTime":"2019-02-21T11:57:28","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
```
Using the info from the occasional exception and the firmware.map the problem is on PubSubClient::publish having not enough stackspace.

Will need to investigate why 6.4.1.14 stackspace is soo much larger than 6.4.1.17 at the time of executing command status 0.

To be continued.

arendst, 2019-02-21:
Caused by lack of stack space it might intermittent go in exception or watchdog. the latest dev frees some stack space and should be more stable.

Give it a try and let me know if it fails again.

DavinKD, 2019-02-21:
I'll try it tonight.  I didn't bring one to work with me today.

DavinKD, 2019-02-24:
Sorry for the slow reply.  I had ordered a few more of these and they just came in today.  That way I didn't have to mess with ones I already had in use.  I just flashed with the latest dev build (6.4.1.18(41a342c-sonoff)).  I am unable to make it crash by running a status.  Looks like we're good now.  Thanks for the fix.

molpie, 2019-02-26:
Hello, two days ago I installed version 6.4.1.19 on Blitzwolf smart sockets. The hardware and software watchdogs that caused the restart have disappeared. I often use TasmoAdmin and don't see any problems.

andrethomas, 2019-02-26:
@molpie Thanks for the feedback. The bug was isolated to around 6.4.1.15-6.4.1.16 and assume stable from 6.4.1.18. Appreciate the feedback.

cmccambridge, 2019-03-28:
@DavinKD From a mechanical side, how did you do your wiring for the SP201? I've got an SP201 that failed OTA conversion, and am trying to work out a reasonable wiring method to access the UART pins on the daughter board... At the moment I think I'll either need some very long probes, or to desolder the AC plug itself if I want to avoid permanently cutting away the back plastic housing. Any tips?

DavinKD, 2019-03-28:
I only used OTA.  I've never tried to open these.

cmccambridge, 2019-03-28:
Oops, I'm sorry about that, thanks for the heads up. I lost track of who was who during the thread. I actually meant to ask @awk2019 for physical access tips based on the comment about:

> tuya-convert on one and cracked on open for wired serial flash.

(FWIW, so far this does not look to be a very physical-modification-friendly device... have to break a glue seal to open it initially, and still TBD what the best serial access route is. OTA on my other one was vastly preferable :) )

ed-war-d, 2019-05-30:
Hi! Could't you tell me where I can find the pinouts (Rx,Tx, 3.3V) for flashing the NX-SP201?

Jason2866, 2019-05-30:
@ed-war-d  if you provide photos probably yes

ed-war-d, 2019-05-30:
[image]
[image]

Jason2866, 2019-05-30:
A photo of the blue Module from the other side will give the needed infos

ed-war-d, 2019-05-30:
[image]

[14 more replies]