arendst/Tasmota #5218: Device doesn't reset after template change in console

blakadder, 2019-02-13:
**Describe the bug**
After changing template by using the built in template numbers in console the device doesn't reset itself.
```

**To Reproduce**
flash fresh firmware .16, 
type `template 3` (or any other template) in console
the console shows changed template but the device doesn't reset

**Expected behavior**
Device should reset to reflect and boot into `module 0` with the new template

this should not be possible
![image](https://user-images.githubusercontent.com/5904370/52735399-d6c8ec00-2fc7-11e9-9700-ed0ef8472dbb.png)

arendst, 2019-02-13:
As designed!

Once selected an inbuilt template while it wasn't the active module (0) itt allows the user to edit it before usage.

If module 0 was active at the time the template command is executed it will restart the device.

arendst, 2019-02-13:
A template is not a module. It is used as one of 70 modules.

andrethomas2, 2019-02-13:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-13 by andrethomas2

blakadder, 2019-02-13:
Ok. I put a warning in the wiki that its necessary to use module 0 for the config to apply

meingraham, 2019-02-13:
So, there is the [Templates documentation](https://github.com/arendst/Sonoff-Tasmota/wiki/Templates) and the [Template.md](https://github.com/arendst/Sonoff-Tasmota/blob/development/TEMPLATE.md). These two need to be merged and one should be deleted? Otherwise updates will likely get quickly out of sync.

I would recommend a separate page for the templates themselves. At the top of that page would be the link to the [templates documentation wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Templates) page.

IMO, the Templates link in the wiki footer should point to the actual Templates and not the Templates doc. The Templates doc page can be linked to from the Commands page, the Initial Configuration page, etc.

Mike

meingraham, 2019-02-14:
I meant to ask... I don't understand BASE. Is there a more detail explanation of this somewhere?

Also, in the wiki documentation example:
[17, | 243, | 29, | 244, | 7, | 201, | 255, | 255, | 138, | 255, | 139, | 255, | 255]

243, 29, 244, 7, 201, 138, & 139... where are these defined?

Also, the documentation has:
9 | Switch1
09 | Relay1

 the '09' must be a typo.

Mike

Jason2866, 2019-02-14:
@meingraham  Step in to Discord :-)  talk to @blakadder  
He is doing a great job writting the wiki entry for "Template".
Every helpful input is welcome.
Thx, we see us @Discord Chat

arendst, 2019-02-14:
@blakadder indeed exactly as I envisioned it. Great job done so far!

blakadder, 2019-02-14:
> I meant to ask... I don't understand BASE. Is there a more detail explanation of this somewhere?
> 
> Also, in the wiki documentation example:
> [17, | 243, | 29, | 244, | 7, | 201, | 255, | 255, | 138, | 255, | 139, | 255, | 255]
> 
> 243, 29, 244, 7, 201, 138, & 139... where are these defined?

https://github.com/arendst/Sonoff-Tasmota/wiki/Templates#gpio-functionality

> Also, the documentation has:
> 9 | Switch1
> 09 | Relay1
> 
> the '09' must be a typo.

That's how it was numbered in the list i got from the code

arendst, 2019-02-14:
Just change the 09 to 21 and all is good to go.

geyseran, 2020-10-17:
Is it possible to connect 2 SR04HC ultrasonic devices to one tasmota device? On the tasmota template, you can only choose one device by selecting SR04 Tri/TX (73) and SR04 Ech/RX (74). It would be nice if one could select more than one device per tasmota device?