arendst/Tasmota #5219: Separate

arjange, 2019-02-13:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

arjange, 2019-02-13:
apparte switch shelly1

Hey,

I am a beginner and I am looking for a way to have my sw connectng on my shelly1 communicate with my domoticz with a separate signal. Now he switches the relai with this and I can not find the right mode. I work with tasmota. Thanks in advance.

baha00, 2019-02-13:
This question is pretty unclear. What are you running on your device? Did you flash it with tasmota? What software is not connecting where??

arjange, 2019-02-14:
Sorry, but my English is not very good. I actually want to disconnect the function of the switch from the function of the relay. After that the shelly1 has a possibility to switch via the relay and the possibility to measure the status of the sw contact. Separate from each other. This has to go to domoticz via Mosqitto but this has already been set up and this is already working. Who helps me ..

arjange, 2019-02-14:
Er draait tasmota 6,4,1 op de shelly1

arendst, 2019-02-14:
Take a look at command ``SwitchTopic``. Search for it in both wiki and issues here. That should solve your issue.

arjange, 2019-02-14:
Thanks a lot. Its working now!

closed 2019-02-14 by arjange