arendst/Tasmota #5220: RFBridge, Tasmota, setting MQTT Full Topic does not work

badabing2005, 2019-02-13:
RFBridge flashed with Tasmota 6.4.1
In the UI change the Full Topic from 
%topic%/%prefix%/
to
%prefix%/%topic%/
save
after it restarts, go back and check
The value is not changed, it's back to 
%topic%/%prefix%/

incidentally, the value in brackets next to Full Topic always reads (%prefix%/%topic%/) not sure why
However if I check the console logs, it's always publishing to %topic%/%prefix%/

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff RFBridge
- [x] _Development/Compiler/Upload tools used_ : released binary uploaded using FlashESP8266.exe
- [x] _If a pre-compiled release or development binary was used, which one?_ : release binary
- [x] _You have tried latest release or development binaries?_ : No
- [x] _Provide the output of command_``status 0`` :
```
18:00:20 CMD: status 0
18:00:20 MQT: rf_bridge2/stat/STATUS = {"Status":{"Module":25,"FriendlyName":["rf_bridge2"],"Topic":"rf_bridge2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
18:00:20 MQT: rf_bridge2/stat/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:24:20","StartupUTC":"2019-02-13T22:36:00","Sleep":50,"BootCount":41,"SaveCount":73,"SaveAddress":"F7000"}}
18:00:20 MQT: rf_bridge2/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
18:00:20 MQT: rf_bridge2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MY_SSID","MY_SSID2"],"TelePeriod":300,"SetOption":["000A8029","558180C0","00000200"]}}
18:00:20 MQT: rf_bridge2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
18:00:20 MQT: rf_bridge2/stat/STATUS5 = {"StatusNET":{"Hostname":"RF_Bridge2","IPAddress":"10.10.10.44","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.10.1","Mac":"REDACTED","Webserver":2,"WifiConfig":5}}
18:00:20 MQT: rf_bridge2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.10.77","MqttPort":1883,"MqttClientMask":"RF_Bridge2","MqttClient":"RF_Bridge2","MqttUser":"hass","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:00:20 MQT: rf_bridge2/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 13 23:00:20 2019","Local":"Wed Feb 13 18:00:20 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"18:01","Sunset":"14:27"}}
18:00:20 MQT: rf_bridge2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-02-13T18:00:20"}}
18:00:20 MQT: rf_bridge2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-02-13T18:00:20","Uptime":"0T00:24:20","Vcc":3.486,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"MY_SSID","BSSId":"REDACTED","Channel":1,"RSSI":62}}}
```

**To Reproduce**
See above

**Expected behavior**
the MQTT settings change to take effect like it does with other devices

**Screenshots**
[image]

**Additional context**
I had also done the RF firmware upgrade mentioned [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433) and flashed RF-Bridge-EFM8BB1-20181127.hex but even reverting to RF_Bridge_iTead_Original.hex did not help

Thanks
BB

jziolkowski, 2019-02-14:
Did you do a full flash erase before flashing?
Also, the values in brackets show default values.

badabing2005, 2019-02-14:
@jziolkowski 
Thanks for responding,
Yes, I did a full flash erase before flashing.
Don't quote me on this, but if I'm not mistaken, the defaults used to be %prefix%/%topic% but at some point that was changed to %topic%/%prefix% (I don't recall at which version) I had to go and change all my devices back to how I had them before.
In fact I didn't realize until today that my changes to RFBridge had not stuck.

I have two RFBridges, and both are acting exactly the same way.

wongnam, 2019-02-14:
@badabing2005 I've just double check for you. all are working normally as expect. prefix was changed after save.

```
00:00:00 Project sonoff RF Bridge 1 Version 6.4.1.15(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 iot-2 in mode 11N as rfbridge1-7677...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on rfbridge1-7677 with IP address 192.168.12.114
11:06:39 MQT: Attempting connection...
11:06:40 MQT: Connected
11:06:40 MQT: rfbridge1/tele/LWT = Online (retained)
11:06:40 MQT: rfbridge1/cmnd/POWER = 
11:06:40 MQT: rfbridge1/tele/INFO1 = {"Module":"Sonoff Bridge","Version":"6.4.1.15(sonoff)","FallbackTopic":"cmnd/DVES_825DFD_fb/","GroupTopic":"sonoffs"}
11:06:40 MQT: rfbridge1/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"rfbridge1-7677","IPAddress":"192.168.12.114"}
11:06:40 MQT: rfbridge1/tele/INFO3 = {"RestartReason":"Software/System restart"}
11:06:48 MQT: rfbridge1/tele/STATE = {"Time":"2019-02-14T11:06:48","Uptime":"0T00:00:14","Vcc":3.486,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":74}}
11:06:48 MQT: rfbridge1/tele/SENSOR = {"Time":"2019-02-14T11:06:48","SI7021":{"Temperature":30.1,"Humidity":54.5},"TempUnit":"C"}
```
screen shot
[image]

wongnam, 2019-02-14:
@badabing2005 Hey, i saw your status 0 show that the full topic was changed as your configuration.
[image]

andrethomas2, 2019-02-14:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-14 by andrethomas2

badabing2005, 2019-02-14:
@wongnam 
How did you derive that it is working?
My topic is rf_bridge2
I want to have the full topic to be **%prefix%**/%topic% (prefix first)
What you are showing me is %topic%/%prefix% (topic first)

I expect
/stat/rf_bridge2
/tele/rf_bridge2
/cmnd/rf_bridge2

even your screenshot shows that you have the same issue.

@arendst unless I'm misunderstanding how this works I believe this was wrongly closed.

For example, my sonoff basic correctly shows this as my mqtt settings (settings get saved)
[image]

instead of this (like in rfbridge)
[image]

and status 0 is also correct
[image]

unlike the rfbridge one.

Thanks
BB

wongnam, 2019-02-14:
@badabing2005 My Sonoff RF bridge have no issue, I can change it any way I want.
Could you try to erase the configuration and try. type command Reset 5 in console.

```
00:00:00 Project sonoff RF Bridge 1 Version 6.4.1.15(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 iot-2 in mode 11N as rfbridge1-7677...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on rfbridge1-7677 with IP address 192.168.12.114
19:54:01 MQT: Attempting connection...
19:54:01 MQT: Connected
19:54:01 MQT: tele/rfbridge1/LWT = Online (retained)
19:54:01 MQT: cmnd/rfbridge1/POWER = 
19:54:01 MQT: tele/rfbridge1/INFO1 = {"Module":"Sonoff Bridge","Version":"6.4.1.15(sonoff)","FallbackTopic":"cmnd/DVES_825DFD_fb/","GroupTopic":"sonoffs"}
19:54:01 MQT: tele/rfbridge1/INFO2 = {"WebServerMode":"Admin","Hostname":"rfbridge1-7677","IPAddress":"192.168.12.114"}
19:54:01 MQT: tele/rfbridge1/INFO3 = {"RestartReason":"Software/System restart"}
19:54:09 MQT: tele/rfbridge1/STATE = {"Time":"2019-02-14T19:54:09","Uptime":"0T00:00:18","Vcc":3.504,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":88}}
19:54:09 MQT: tele/rfbridge1/SENSOR = {"Time":"2019-02-14T19:54:09","SI7021":{"Temperature":32.5,"Humidity":51.4},"TempUnit":"C"}
```
[image]

[image]

wongnam, 2019-02-14:
@badabing2005 Please make sure that the configuration is saved, sometimes the "save button" is obscured 
my screen shot
[image]

jziolkowski, 2019-02-14:
check if you don't have SetOption19 enabled. It is for Home Assistant discovery and it flips the topic around. Tasmota default is %prefix%/%topic%/

badabing2005, 2019-02-15:
Thank you @wongnam  and @jziolkowski 

> check if you don't have SetOption19 enabled. It is for Home Assistant discovery and it flips the topic around. Tasmota default is %prefix%/%topic%/

22:18:33 CMD: setoption 19
22:18:33 MQT: rf_bridge2/stat/RESULT = {"SetOption1":"OFF"}

> Could you try to erase the configuration and try. type command Reset 5 in console.

That indeed fixed the issue. Thank you very much.

On one of the devices, the MQTT settings were erased after the reset 5
whereas on the other it was retained.
I wonder why.
In any case both devices have now the correct Full Topic.

Thanks