arendst/Tasmota #5221: 4CH Pro R2 with External Switches

speedyrazor, 2019-02-14:
I have the 4CH Pro R2, where button 1 is not an issue to connect external switch to.
I just tested with wiring a switch direct to each button, and although it works, it's very slow to respond to the switch (maybe 700ms or more, not instant like it is when you assign switch1 to RX for example). I see someone in another thread mentioning GPIO_USER (looks interesting), but my I don't quite understand. Could someone elaborate on the best (easy'ish) way to add external switches to all 4 buttons of Sonoff 4CH Pro R2 please?

ascillato, 2019-02-14:
Hi,

Please, search in the wiki for setoption1 and setoption13 to reduce that time disabling multipress feature. Thanks.

speedyrazor, 2019-02-14:
Hi, the default value of of SetOption1 and SetOption13 is 'off', I have tried variations of 'on' and 'off', but the switch is still at least 700ms or more, or am I misunderstanding something?

ascillato2, 2019-02-15:
Ok, something weird is happening. For further Support, please complete the troubleshooting template and add the output of the console when this happens. Please use weblog 4. Or if you prefer address this to the Tasmota Support Chat.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-15 by ascillato2