arendst/Tasmota #5222: Custom Template and WebUI GPIO display

blakadder, 2019-02-14:
It might be the lack of time spent but afaik when I activate a custom template the WebUI module configuration page shows all the defined GPIO pins with a dropdown menu although none of the pins is set to 255. 
That might cause user confusion and raised issues. 

I propose to either hide the templated gpios showing only 255 user defined one as in the all other supported modules. 
While at the topic is it possible to have the webui not show the recommended pin assignments when module 0 is selected to also reduce user confusion (just like module 18 adds D# pin information, remove any information apart from GPIO#)

arendst, 2019-02-14:
Showing template GPIOs was a design decision I made. I will re-consider.

Removing the recommended pin assignments is an option to consider too but as the user template is just another template I think it makes sense to have the same info as the inbuilt templates.

arendst, 2019-02-14:
Ah, just did some tests and I now know why I did it that way. 

As the template is updated with user defined GPIOs the next time the module page is pulled up it will not allow to make a changes to a now defined GPIO.

I will have to dig deeper. WIll roll back the latest change.

Jason2866, 2019-02-14:
Mhh, i liked the way before this change to create Templates via WebConfig
It was possible to config in Module generic (0) all the GPIOs you want need.
After that just typing in WebConsole `Template 0` and i got the Template config.
Now this isnt possible anymore i always get 255...

arendst, 2019-02-14:
Then when you re-visit the module web page the template was changed with the latest module changes. I agreed with @blakadder that it might confuse users.

That's why I started to come up with a new menu:

[image]

Work in progress but this allows to change the template online without the user getting confused about a dynamic template.

So pls hold on, I'm working on it but I need time to cram it all in no code space.

Jason2866, 2019-02-14:
Ahh, thx
Had the fear to loose the nice possibility to do template config via webfrontend.
**No rush** if this feature will be there again :-)

blakadder, 2019-02-15:
Yep, that looks more usable and very straightforward for a majority of users

arendst, 2019-02-17:
Initial released today. Working on tuning.

Jason2866, 2019-02-17:
If i define a template and use it afterwards the defined GPIOS are not visible.
Is it the way you want, or what you are going to tune?
[image]

[image]

arendst, 2019-02-17:
This is the way I want it to be.

As in embedded templates, defined GPIOs are not visible in the module menu and cannot be changed by the user.

Jason2866, 2019-02-17:
Thx, for explaining

blakadder, 2019-02-17:
[image]

Device doesn't restart after clicking save template

blakadder, 2019-02-17:
[image]
Device doesn't restart after clicking save in Configure Template

wongnam, 2019-02-17:
> Device doesn't restart after clicking save in Configure Template
Confirm, I also encounter this issue. It will be happen when i change the name.

[image]

lz2ddl, 2019-02-18:
I can not add new bulbs (for example, 4 or 5) to Hue Bridge. What to do?

arendst, 2019-02-18:
Only one WS2812 strip is supported.

lz2ddl, 2019-02-18:
A normal RGB strip with PCA9685?

ascillato, 2019-02-18:
@lz2ddl

Sorry, you are offtopic of the current thread. Please, address your questions to the Tasmota Support Chat. Will be better and more dynamic. Thanks. 

See readme for the link to the chat.

arendst, 2019-02-18:
In that case open a new issue dedicated to your problem as this is another issue.

closed 2019-02-18 by arendst