arendst/Tasmota #5223: Complete GPIOS list

blakadder, 2019-02-14:
Can you provide a definitive GPIOS list for Template since there are still some missing in the console command.

arendst, 2019-02-14:
You'll find the references as an enumerated list in sonoff_template.h 
```
  GPIO_DHT11,          // DHT11
  GPIO_DHT22,          // DHT21, DHT22, AM2301, AM2302, AM2321
  GPIO_SI7021,         // iTead SI7021
  GPIO_DSB,            // Single wire DS18B20 or DS18S20
  GPIO_I2C_SCL,        // I2C SCL
  GPIO_I2C_SDA,        // I2C SDA
  GPIO_WS2812,         // WS2812 Led string
  GPIO_IRSEND,         // IR remote
  GPIO_SWT1,           // User connected external switches
  GPIO_SWT2,
  GPIO_SWT3,
  GPIO_SWT4,
  GPIO_SWT5,
  GPIO_SWT6,
  GPIO_SWT7,
  GPIO_SWT8,
  GPIO_KEY1,           // Button usually connected to GPIO0
  GPIO_KEY2,
  GPIO_KEY3,
  GPIO_KEY4,
  GPIO_REL1,           // Relays
  GPIO_REL2,
  GPIO_REL3,
  GPIO_REL4,
  GPIO_REL5,
  GPIO_REL6,
  GPIO_REL7,
  GPIO_REL8,
  GPIO_REL1_INV,
  GPIO_REL2_INV,
  GPIO_REL3_INV,
  GPIO_REL4_INV,
  GPIO_REL5_INV,
  GPIO_REL6_INV,
  GPIO_REL7_INV,
  GPIO_REL8_INV,
  GPIO_PWM1,           // RGB   Red   or C  Cold White
  GPIO_PWM2,           // RGB   Green or CW Warm White
  GPIO_PWM3,           // RGB   Blue
  GPIO_PWM4,           // RGBW  (Cold) White
  GPIO_PWM5,           // RGBCW Warm White
  GPIO_CNTR1,
  GPIO_CNTR2,
  GPIO_CNTR3,
  GPIO_CNTR4,
  GPIO_PWM1_INV,       // RGB   Red   or C  Cold White
  GPIO_PWM2_INV,       // RGB   Green or CW Warm White
  GPIO_PWM3_INV,       // RGB   Blue
  GPIO_PWM4_INV,       // RGBW  (Cold) White
  GPIO_PWM5_INV,       // RGBCW Warm White
  GPIO_IRRECV,         // IR receiver
  GPIO_LED1,           // Leds
  GPIO_LED2,
  GPIO_LED3,
  GPIO_LED4,
  GPIO_LED1_INV,
  GPIO_LED2_INV,
  GPIO_LED3_INV,
  GPIO_LED4_INV,
  GPIO_MHZ_TXD,        // MH-Z19 Serial interface
  GPIO_MHZ_RXD,        // MH-Z19 Serial interface
  GPIO_PZEM0XX_TX,     // PZEM0XX Serial interface
  GPIO_PZEM004_RX,     // PZEM004T Serial interface
  GPIO_SAIR_TX,        // SenseAir Serial interface
  GPIO_SAIR_RX,        // SenseAir Serial interface
  GPIO_SPI_CS,         // SPI Chip Select
  GPIO_SPI_DC,         // SPI Data Direction
  GPIO_BACKLIGHT,      // Display backlight control
[... 202 lines cut ...]
#define D_SENSOR_MAX31855_CLK  "MX31855 CLK"
#define D_SENSOR_MAX31855_DO   "MX31855 DO"
#define D_SENSOR_NRG_SEL       "HLWBL SEL"  // Suffix "i"
#define D_SENSOR_NRG_CF1       "HLWBL CF1"
#define D_SENSOR_HLW_CF        "HLW8012 CF"
#define D_SENSOR_HJL_CF        "BL0937 CF"
#define D_SENSOR_MCP39F5_TX    "MCP39F5 Tx"
#define D_SENSOR_MCP39F5_RX    "MCP39F5 Rx"
#define D_SENSOR_MCP39F5_RST   "MCP39F5 Rst"
#define D_SENSOR_CSE7766_TX    "CSE7766 Tx"
#define D_SENSOR_CSE7766_RX    "CSE7766 Rx"
#define D_SENSOR_PN532_TX      "PN532 Tx"
#define D_SENSOR_PN532_RX      "PN532 Rx"
#define D_SENSOR_SM16716_CLK   "SM16716 CLK"
#define D_SENSOR_SM16716_DAT   "SM16716 DAT"
#define D_SENSOR_SM16716_POWER "SM16716 PWR"
#define D_SENSOR_MY92X1_DI     "MY92x1 DI"
#define D_SENSOR_MY92X1_DCKI   "MY92x1 DCKI"
#define D_SENSOR_ARIRFRCV      "ALux IrRcv"
#define D_SENSOR_TXD           "Serial Tx"
#define D_SENSOR_RXD           "Serial Rx"
#define D_SENSOR_ROTARY        "Rotary"     // Suffix "1A"
```

I hope that provides the info you're looking for.

blakadder, 2019-02-14:
still missing  243, 244, 201 that are used in your Template example

arendst, 2019-02-14:
S**t. They are wrong and shouldn't be there. I was just looking at it in my wemos and thought where do they come from...

Change them as follows:
243 = 148
244 = 149
201 = 255

closed 2019-02-15 by ascillato2