arendst/Tasmota #5225: Timer and rule problem in sonoff bridge

solucionescol, 2019-02-14:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Development/Compiler/Upload tools used :
- [ x] Hardware used : Sonoff bridge
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ x] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**
hi

i have the following rule for timer1

`on clock#timer=1 do mem2 on endon`

The timer and this rule works ok

i have this another rule

`on mem2#State do publish stat/test/stat endon`

this rule is ON and is not working. (nothing is pubished to mqtt), i tried with another actions and nothing is executed.

am i missing something?

thankyou

ascillato, 2019-02-14:
Hi,

Your template is incomplete so a lot of information is missing. The mem#state was recently added and without the template we don't know if you are using latest or not. If not, please update. If you are using latest version (6.4.1.16) please, complete the troubleshooting template. Thanks

solucionescol, 2019-02-14:
You are right, updated and solved.

thankyou.

closed 2019-02-14 by solucionescol