arendst/Tasmota #5226: Sonoff T1 433 Mhz Feature / reconfigure or disable?

unimatrix27, 2019-02-14:
The T1 has a 433 Mhz function that is basically supported by Tasmota as also stated in the WIKI. However I am using the T1 also for long press button actions, and this very often leads to the user pressing the button for 4 seconds and thus giving the "beep" and put the T1 into 433 Pairing Mode. 

Is there any way to:

- Independently parse the remote commands received via 433 Mhz independently from the relays?
- Disable paring feature completely to not accidently use it (i am having a lot of PIR devices sending on 433 mhz so they could be accidently paired)

ascillato2, 2019-02-15:
The pairing is done by the rf chip, not Tasmota. So, it can't be disabled.
The button is connected to both (esp8266 and rf chip), so, it isn't recommended to use hold feature on that device. 
The rf chip "press" the button when receives the rf signal. So, the only way to disable that isa to modify the template and set gpio0 as a switch and use a rule to catch it. 

You can check the wiki on how to do that. Anyway, if you need assistant please do not hesitate on asking here or in the Tasmota Support Chat. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-15 by ascillato2

unimatrix27, 2019-02-15:
I am wondering if the RF chip could be flashed with alternative firmware like its possible for the [RF Bridge](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433).  I would be happy if at least the time could be extended from 4 seconds to something like 10 seconds.