arendst/Tasmota #5228: Sonoff Ghost Switching Correct Retain Settings

yasas100, 2019-02-14:
My Sonoff randomly switches on when it disconnects from the wifi. I believe my setting are correct in HA. It is connected to a switch and I am even using a low pass filter to avoid hardware ghosting. I have posted screenshots from my console below. Thanks in advance

MQT: Attempting connection...

20:05:53 MQT: Connected

20:05:53 MQT: tele/sonoff1/LWT = Online (retained)

20:05:53 MQT: cmnd/sonoff1/POWER =

20:05:53 MQT: stat/sonoff1/RESULT = {"POWER":"ON"}

20:05:53 MQT: stat/sonoff1/POWER = ON (retained)

[image]

[image]

[image]

ascillato2, 2019-02-15:
Hi,

You have retained messages in your broker. You need to delete the broker database so as to solve your issue. 

If you need further assistant with that, please find us in the Tasmota Support Chat. Thanks. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-15 by ascillato2

HobbytronicsPK, 2019-05-13:
Have a look on to [Ghost / Random Switching on Sonoff Devices (MQTT Retained)](https://blog.hobbytronics.pk/sonoff-tasmota-tips-tricks/#Ghost-Random-Switching)