arendst/Tasmota #5229: TSL2541 and BME280 on NodeMCU no MQTT only RSL

stephan1980, 2019-02-14:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE 1.8.5 Mac OSX
- [x] Hardware used : Amica Node MCU
- [ ] If a pre-compiled release or development binary was used, which one? : Tasmota 6.4.1.17
- [x] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
00:23:32 CMD: Status 0
00:23:32 RSL: /SmartHome/Wettersensor/STATUS = {"Status":{"Module":18,"FriendlyName":["Wettersensor"],"Topic":"Wettersensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:28","StartupUTC":"2019-02-14T23:16:04","Sleep":50,"CfgHolder":4617,"BootCount":49,"SaveCount":66,"SaveAddress":"3FB000"}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(sonoff)","BuildDateTime":"2019.02.15 00:04:03","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":3,"LogHost":"","LogPort":514,"SSId":["Solaris",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS4 = {"StatusMEM":{"ProgramSize":453,"Free":2616,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":2,"Features":["00000407","0F022794","00000000","000804C6","000000C0"]}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS5 = {"StatusNET":{"Hostname":"Wettersensor-2952","IPAddress":"192.168.60.155","Gateway":"192.168.60.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.60.1","Mac":"EC:FA:BC:05:8B:88","Webserver":2,"WifiConfig":4}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.60.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_058B88","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 14 23:23:32 2019","Local":"Fri Feb 15 00:23:32 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:58","Sunset":"18:10"}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS10 = {"StatusSNS":{"Time":"2019-02-15T00:23:32","BME280":{"Temperature":21.5,"Humidity":41.6,"Pressure":1030.7},"TSL2561":{"Illuminance":11.270},"PressureUnit":"hPa","TempUnit":"C"}}
00:23:32 RSL: /SmartHome/Wettersensor/STATUS11 = {"StatusSTS":{"Time":"2019-02-15T00:23:32","Uptime":"0T00:07:28","Vcc":2.796,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":100}}}
00:23:35 RSL: /SmartHome/Wettersensor/STATE = {"Time":"2019-02-15T00:23:35","Uptime":"0T00:07:31","Vcc":2.798,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":100}}
00:23:35 RSL: /SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-15T00:23:35","BME280":{"Temperature":21.5,"Humidity":41.7,"Pressure":1030.7},"TSL2561":{"Illuminance":11.670},"PressureUnit":"hPa","TempUnit":"C"}
00:23:45 RSL: /SmartHome/Wettersensor/STATE = {"Time":"2019-02-15T00:23:44","Uptime":"0T00:07:40","Vcc":2.796,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":88}}
00:23:45 RSL: /SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-15T00:23:45","BME280":{"Temperature":21.6,"Humidity":41.5,"Pressure":1030.7},"TSL2561":{"Illuminance":10.810},"PressureUnit":"hPa","TempUnit":"C"}
00:23:55 RSL: /SmartHome/Wettersensor/STATE = {"Time":"2019-02-15T00:23:55","Uptime":"0T00:07:51","Vcc":2.796,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":84}}
00:23:55 RSL: /SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-15T00:23:55","BME280":{"Temperature":21.6,"Humidity":42.6,"Pressure":1030.8},"TSL2561":{"Illuminance":10.840},"PressureUnit":"hPa","TempUnit":"C"}

Hi there,
since i am using several Tasmota devices I am not completely new to it. 
I just build a new project using a TSL2561 and a BME280. Both wired to the D1 and D2 of my NodeMCU. 
When I only connect the TSL2561 the device sends the SENSOR values via MQTT to my broker. 

00:37:34 RSL: /SmartHome/Wettersensor/STATE = {"Time":"2019-02-15T00:37:34","Uptime":"0T00:00:17","Vcc":2.814,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":100}}
00:37:34 MQT: /SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-15T00:37:34","TSL2561":{"Illuminance":8.174}}

Everything works fine.  When I now connect the BME280 instead the SENSOR data itsn't submitted via MQTT. The console only shows the RSL not the MQT. Same is happening when I connect both sensors.

00:36:27 RSL: /SmartHome/Wettersensor/STATE = {"Time":"2019-02-15T00:36:27","Uptime":"0T00:20:23","Vcc":2.800,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Solaris","BSSId":"44:4E:6D:63:34:8C","Channel":1,"RSSI":100}}
00:36:27 RSL: /SmartHome/Wettersensor/SENSOR = {"Time":"2019-02-15T00:36:27","BME280":{"Temperature":21.9,"Humidity":40.4,"Pressure":1030.7},"TSL2561":{"Illuminance":8.469},"PressureUnit":"hPa","TempUnit":"C"}

The webinterface displays both sensors correctly.

![bildschirmfoto 2019-02-15 um 00 33 43](https://user-images.githubusercontent.com/39016168/52824739-12dc7980-30ba-11e9-84ea-f1f52a7f6122.png)

My broker is working fine, since I am receiving multiple sensor data from other devices
Anything I could try or do?

Best regards
Stephan

Jason2866, 2019-02-15:
It seems your self compiled firmware is faulty. Try precompiled (sonoff-sensors.bin) from 
http://thehackbox.org/tasmota/
I use many BME280 and there is no problem with.

stephan1980, 2019-02-16:
Thank you Jason, that worked for me. It seems that my firmware is somehow faulty....

closed 2019-02-16 by stephan1980