arendst/Tasmota #5232: Working device - eMylo SS-8839-02

sqerk, 2019-02-15:
I would just like to add that these devices work.
Used https://github.com/arendst/Sonoff-Tasmota/wiki/CE-Smart-Home---LA-WF3-Wifi-Plug-(TYWE2S) for the soldering guide.
But no need to change settings flashed the same firmware and work the same as Sonoff S31

blakadder, 2019-02-15:
you're free to add the information in the wiki, preferably with an entry in **Other devices** which links to the page you posted which you can expand with the name and link to the device

ascillato2, 2019-02-15:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-15 by ascillato2

andylebedev, 2019-06-21:
not working at all with s31 template, for me it works with this template https://blakadder.github.io/templates/emylo_XL9251WI.html