arendst/Tasmota #5233: 4CH Pro R2 with External Switches - Slow response

speedyrazor, 2019-02-15:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Development/Compiler/Upload tools used :
- [x ] Hardware used :  4CH Pro R2 + External rocker switch
- [x ] If a pre-compiled release or development binary was used, which one? : Sonoff-Tasmota v6.4.1
- [x ] You have tried latest release or development binaries? : no
- [x ] Provide the output of command ``status 0`` : yes
```
07:23:17 CMD: status 0
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff1","Sonoff2","Sonoff3","Sonoff4"],"Topic":"Sonoff_4CH_Pro","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:01:22","StartupUTC":"2019-02-15T07:21:55","Sleep":50,"BootCount":33,"SaveCount":392,"SaveAddress":"F4000"}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["KevinWifi",""],"TelePeriod":300,"SetOption":["0000A009","55A180C0","00000000"]}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS5 = {"StatusNET":{"Hostname":"Sonoff_4CH_Pro","IPAddress":"192.168.0.23","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:DB:FE:C3","Webserver":2,"WifiConfig":4}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.50","MqttPort":1883,"MqttClientMask":"Sonoff_4CH_Pro","MqttClient":"Sonoff_4CH_Pro","MqttUser":"kevin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 15 07:23:17 2019","Local":"Fri Feb 15 07:23:17 2019","StartDST":"Sun Mar 31 01:00:00 2019","EndDST":"Sun Oct 27 02:00:00 2019","Timezone":99,"Sunrise":"06:58","Sunset":"17:10"}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS10 = {"StatusSNS":{"Time":"2019-02-15T07:23:17"}}
07:23:17 MQT: stat/Sonoff_4CH_Pro/STATUS11 = {"StatusSTS":{"Time":"2019-02-15T07:23:17","Uptime":"0T00:01:22","Vcc":3.480,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"KevinWifi","BSSId":"8C:3B:AD:30:88:F1","Channel":8,"RSSI":82}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Hi, I have the 4CH Pro R2, where button 1 is not an issue to connect external switch to.
I just tested with wiring a switch direct to each button, and although it works, it's very slow to respond to the switch (maybe 700ms or more, not instant like it is when you assign switch1 to RX for example). The default value of of SetOption1 and SetOption13 is 'off', I have tried variations of 'on' and 'off', but when using the external switch there is still at least 700ms or more delay.
Here is the console output when using the external switch on switch 4:

07:25:25 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"ON"}
07:25:25 MQT: stat/Sonoff_4CH_Pro/POWER4 = ON
07:25:28 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"OFF"}
07:25:28 MQT: stat/Sonoff_4CH_Pro/POWER4 = OFF

Kind regards.

ascillato, 2019-02-15:
Hi,

As you were asked in your other issue, please share the output of the console when switching but using weblog 4.

The 2 setoption explained are the ones that controls the timing. So, it is a hardware issue or it is a wifi disconnections. We can see that from the full console log. Please, share it. Thanks.

speedyrazor, 2019-02-15:
Sorry, I missed that part of the request, please see below for the console output with weblog 4:

```
12:43:41 HTP: WeMo basic event
12:43:44 APP: Button4 level 0-1
12:43:44 SRC: Button
12:43:44 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"OFF"}
12:43:44 MQT: stat/Sonoff_4CH_Pro/POWER4 = OFF
12:43:45 CFG: Saved to flash at F7, Count 413, Bytes 3584
12:43:51 HTP: WeMo basic event
12:43:53 APP: Button4 level 1-0
12:43:53 SRC: Button
12:43:53 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"ON"}
12:43:53 MQT: stat/Sonoff_4CH_Pro/POWER4 = ON
12:43:54 CFG: Saved to flash at F6, Count 414, Bytes 3584
12:44:00 WIF: Checking connection...
12:44:00 WIF: Connected
12:44:02 HTP: WeMo basic event
```

ascillato, 2019-02-15:
Hi, please you try to disable emulation to test the speed response again.

speedyrazor, 2019-02-15:
Hi, this is with emulation disabled:

```
14:09:25 APP: Button4 level 0-1
14:09:25 SRC: Button
14:09:25 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"OFF"}
14:09:25 MQT: stat/Sonoff_4CH_Pro/POWER4 = OFF
14:09:26 CFG: Saved to flash at F8, Count 452, Bytes 3584
14:09:28 WIF: Checking connection...
14:09:28 WIF: Connected
14:09:29 APP: Button4 level 1-0
14:09:29 SRC: Button
14:09:29 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"ON"}
14:09:29 MQT: stat/Sonoff_4CH_Pro/POWER4 = ON
14:09:29 CFG: Saved to flash at F7, Count 453, Bytes 3584
14:09:34 APP: Button4 level 0-1
14:09:34 SRC: Button
14:09:34 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"OFF"}
14:09:34 MQT: stat/Sonoff_4CH_Pro/POWER4 = OFF
14:09:35 CFG: Saved to flash at F6, Count 454, Bytes 3584
14:09:37 APP: Button4 level 1-0
14:09:37 SRC: Button
14:09:37 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"ON"}
14:09:37 MQT: stat/Sonoff_4CH_Pro/POWER4 = ON
14:09:37 APP: Button4 level 0-1
14:09:38 CFG: Saved to flash at F5, Count 455, Bytes 3584
14:09:48 WIF: Checking connection...
14:09:48 WIF: Connected
14:09:57 APP: Button4 level 1-0
14:09:57 SRC: Button
14:09:57 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"OFF"}
14:09:57 MQT: stat/Sonoff_4CH_Pro/POWER4 = OFF
14:09:58 CFG: Saved to flash at F4, Count 456, Bytes 3584
14:10:00 APP: Button4 level 0-1
14:10:00 SRC: Button
14:10:00 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"ON"}
14:10:00 MQT: stat/Sonoff_4CH_Pro/POWER4 = ON
14:10:01 CFG: Saved to flash at FB, Count 457, Bytes 3584
14:10:06 APP: Button4 level 1-0
14:10:06 SRC: Button
14:10:06 MQT: stat/Sonoff_4CH_Pro/RESULT = {"POWER4":"OFF"}
14:10:06 MQT: stat/Sonoff_4CH_Pro/POWER4 = OFF
14:10:06 CFG: Saved to flash at FA, Count 458, Bytes 3584
14:10:08 WIF: Checking connection...
14:10:08 WIF: Connected
```

ascillato, 2019-02-15:
And the speed response is also laggy?

speedyrazor, 2019-02-15:
Yes, still the same.

ascillato, 2019-02-15:
Please, try http://thehackbox.org/tasmota/020500/sonoff.bin and then do a `reset 5` in the console

speedyrazor, 2019-02-15:
thanks for the new version. I have installed that as per your above instructions and re-run the commands:

```
18:48:51 CMD: status 0
18:48:51 SRC: WebConsole from 192.168.0.13
18:48:51 RSL: Received Topic /status, Data Size 1, Data 0
18:48:51 RSL: Group 0, Index 1, Command STATUS, Data 0
18:48:51 MQT: stat/sonoff/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:48:51 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:01:22","StartupUTC":"2019-02-15T17:47:29","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":52,"SaveAddress":"F8000"}}
18:48:51 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(b4b1ee9-sonoff)","BuildDateTime":"2019-02-14T18:11:44","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
18:48:51 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["KevinWifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","280500000100000000000000000000000000","00000000"]}}
18:48:51 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":556,"Free":444,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000783A0","23B617CE","00003BC0"]}}
18:48:51 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7875","IPAddress":"192.168.0.14","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:DB:FE:C3","Webserver":2,"WifiConfig":4}}
18:48:51 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DBFEC3C3","MqttUser":"kevin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:48:51 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 15 17:48:51 2019","Local":"Fri Feb 15 18:48:51 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:58","Sunset":"18:10"}}
18:48:52 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-15T18:48:51"}}
18:48:52 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-15T18:48:52","Uptime":"0T00:01:23","Vcc":3.480,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"KevinWifi","BSSId":"8C:3B:AD:30:88:F1","Channel":13,"RSSI":78}}}
```

And then this:

```
18:49:28 APP: Button4 level 1-0
18:49:28 SRC: Button
18:49:28 MQT: stat/sonoff/RESULT = {"POWER4":"ON"}
18:49:28 MQT: stat/sonoff/POWER4 = ON
18:49:28 CFG: Saved to flash at F7, Count 53, Bytes 3584
18:49:28 APP: Button4 level 0-1
18:49:32 APP: Button4 level 1-0
18:49:32 SRC: Button
18:49:32 MQT: stat/sonoff/RESULT = {"POWER4":"OFF"}
18:49:32 MQT: stat/sonoff/POWER4 = OFF
18:49:32 APP: Button4 level 0-1
18:49:32 CFG: Saved to flash at F6, Count 54, Bytes 3584
18:49:35 WIF: Checking connection...
18:49:35 WIF: Connected
18:49:37 APP: Button4 level 1-0
18:49:37 SRC: Button
18:49:37 MQT: stat/sonoff/RESULT = {"POWER4":"ON"}
18:49:37 MQT: stat/sonoff/POWER4 = ON
18:49:37 CFG: Saved to flash at F5, Count 55, Bytes 3584
18:49:37 APP: Button4 level 0-1
18:49:40 APP: Button4 level 1-0
18:49:40 SRC: Button
18:49:40 MQT: stat/sonoff/RESULT = {"POWER4":"OFF"}
18:49:40 MQT: stat/sonoff/POWER4 = OFF
18:49:40 APP: Button4 level 0-1
18:49:40 CFG: Saved to flash at F4, Count 56, Bytes 3584
```

Thanks

ascillato2, 2019-02-24:
Hi,

Any news on this? 

As I could not reproduce this issue, I don't know what else we can try. 

In my devices the response is instantly. 

Do you have any extra capacitor with the external switch?

speedyrazor, 2019-02-24:
Hi, I have no extra capacitor with the external switch.
Could you describe, and or show pictures, of exactly what you are doing to use an external switch on the push button pins to get instant action please? So I can check from what Tasmota version, what console commands, how you wiring and switch looks, and the reaction time.
Kind regards.

closed 2019-03-26 by stale[bot]

bjschuit, 2020-07-06:
I have the same issue. I've got exactly the same 230V motors on all my blinds. 4 of them are being controlled by a Sonoff CH4 Pro R2, one of them with a Sonoff Dual. When either controlled via Home Assistant via mqtt or web interface, the Dual responds instantly, while the 4CH Pro R2 has some delay. It is not a physical delay, as you here the relays switch with some delay after pressing a button.