arendst/Tasmota #5234: Wrong wifi password causes loop

spock64, 2019-02-15:
There is a problem where a brand new unit is flashed via OTA, presents its AP and Web configuration UI on restart, but fails to connect to the Wifi (due to password entry bugs elsewhere) but does not, after a timeout, start the AP and Web UI again. Because the Unit is closed (a mains plug glued shut), this means that recovering form this error is time consuming.

I suspect that the reason that the AP / WEB UI is not started is due to the Wifi router not responding, causing the timeout messages (below) and the code not treating this error as being analogous to an incorrect password.

Sequence:
1. New unit flashed with Sonoff.bin
2. Password entered via web UI, unit restarts.
3. Unit fails to connect (due to the password being stored incorrectly - see bugs including #5231 amongst others) but the unit never starts the web configuration AP. 

Debug output is:

`00:07:35 WIF: Connecting to AP1 ROGERS in mode 11N as sonoff-3959...

00:07:52 WIF: Connect failed with AP timeout

00:07:52 WIF: Connecting to AP1 ROGERS in mode 11N as sonoff-3959...

00:08:09 WIF: Connect failed with AP timeout`

These messages repeat without the AP / WEB UI being started.

[ x] _Searched the problem in issues and in the wiki_
[ x] _Hardware used_ : Wemos D1 plus other custom board.
[ x] _Development/Compiler/Upload tools used_ : See status 0 output below
[ x] _If a pre-compiled release or development binary was used, which one?_ : RELEASE and self built binaries used
[ x] _You have tried latest release or development binaries?_ : YES
[ x] _Provide the output of command_``status 0`` :

```
00:17:40 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:17:40 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:17:40","StartupUTC":"","Sleep":50,"BootCount":11,"SaveCount":17,"SaveAddress":"FB000"}}
00:17:40 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-02-15T10:29:17","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:17:40 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ROGERS",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
00:17:40 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":518,"Free":484,"Heap":30,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE794","240383A0","23B617CE","00003BC0"]}}
00:17:40 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3959","IPAddress":"0.0.0.0","Gateway":"192.168.16.1","Subnetmask":"255.255.252.0","DNSServer":"192.168.16.1","Mac":"68:C6:3A:89:EF:77","Webserver":2,"WifiConfig":4}}
00:17:40 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_89EF77","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:17:40 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:17:40 1970","Local":"Thu Jan 01 00:17:40 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:17:40 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:17:40"}}
00:17:40 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:17:40","Uptime":"0T00:17:40","Vcc":2.994,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ROGERS","BSSId":"FF:FF:FF:FF:FF:FF","Channel":9,"RSSI":68}}}
```

**To Reproduce**
Enter incorrect password in Web UI. On restart, unit fails to connect.

**Expected behavior**
Any failure to connect WiFi should result in the WEB UI and AP being started.

ascillato, 2019-02-15:
Hi,

That is a design decision. If the ssid is blank, it goes to the AP mode. If the ssid exist, wificonfig 4 will retry the connection. 
For your case, the solution is to press 4 times in a row the button connected to gpio0 of your D1 mini to force entering in AP mode. 

Another option is to press for 40 seconds to full reset the device. You should use the one that it is better for you.

Thanks

ascillato2, 2019-02-15:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-15 by ascillato2

spock64, 2019-02-15:
You've closed this a little hastily I think!

The SSID is not blank - you can see that from the quoted debug output. The AP is live - it is simply choosing not to respond when the password is wrong / corrupt. The hardware does not have an external button on GPIO 0. The only way to recover it is to open up the (glued) case and start soldering wires.

If the router responded with incorrect password, the AP would be started. I think it is a design improvement to start the AP timeout from the wifi connection attempt (or maybe actually a bug in error recovery)

If nothing is done about this, we will end up "bricking" perfectly good units!

ascillato, 2019-02-15:
Hi,

I'm sorry. Seems that I didn't explain it well. 

Tasmota can be configured at compiling time and also at runtime which behaviour you want for wifi management. 

At compile time is the key called wificonfig in my_user_config.h 
At runtime is the console command wificonfig 

If you check the wiki at commands, you can configure to go to AP if it can't connect. 

By default, Tasmota uses wificonfig = wifi_retry: If the SSID is configured, it will try to connect to it.

As Tasmota is meant for sonoff devices, you can go into AP mode by just pressing 4 times the button attached to gpio0 

As your case is not standard due you have made a case with no access to the onboard button in a custom board, you should compile by yourself using other wificonfig option that fit your use-case.

Thanks.

spock64, 2019-02-15:
Thanks Adrian - that’s helpful.

I’m not sure why they ended up putting the button on GPIO 13, but that’s hardware guys for you :-)

I’ll also make a fix for the (in) famous password problem sometime this afternoon.

Cheers
Jason

Sent from my phone ...

CindaEames, 2020-03-04:
If you don't know router's default password, here is the list:
[https://www.192-168-0-1.us/default-router-username-password-list/](url)