arendst/Tasmota #5236: How ZMPT101B AC Voltage Sensor module works on Tasmota (?)

adharyanto, 2019-02-15:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-02-15:
Not supported

adharyanto, 2019-02-15:
Will it be supported (planned to be) in the next release of Tasmota?

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

adharyanto, 2019-02-15:
Will it be supported (planned to be) in the next release of Tasmota?

andrethomas, 2019-02-15:
> Will it be supported (planned to be) in the next release of Tasmota?

Most likely one of the developers active on this project will at some point in the future add support for it but for now most people are just using the PZEM modules.

Don't hold your breath though... serious risk of suffocation.

ascillato2, 2019-02-15:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-15 by ascillato2

Zodiark99BR, 2023-08-22:
anything aboout this case ?

barbudor, 2023-08-22:
This is an analog sensor
It can be directly supported using an ADC gpio in ADC Range mode with proper Adcparams

If you want help, open a discussion.
Issues are not for that