arendst/Tasmota #5237: How to poll a switch status?

jacoscar, 2019-02-15:
I have a sonoff basic flashed with tasmota and I’ve connected a door sensor following the instructions from this video https://youtu.be/qZqmkhpJ55A
It works well apart from when the state change happens during an Home Assistant/mqtt restart
If there is a state change of the sensor during those couple of minutes, the state shown in HA is wrong until the state changes again
To fix this it was suggested to poll the state of the sensor upon HA start by using sonoffs/cmnd/state without a payload

13:59:05 CMD: sonoffs/cmnd/state
13:59:05 MQT: stat/sonoff/RESULT = {"Time":"2019-02-13T13:59:05","Uptime":"0T05:04:58","Vcc":3.416,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"...........}

As you can see it reports the status of POWER, but not of POWER2
I also tried sonoffs/cmnd/POWER2 but the response is ‘command unknown’

andrethomas, 2019-02-15:
You may send a command to the topic of your module + status with payload of 10, i.e.

`topic: cmnd/sonoff/status`

`payload: 10`

Response will some something like

`
stat/dining1/STATUS10 = {"StatusSNS":{"Time":"2019-02-15T15:22:30","Switch1":"ON","SI7020":{"Temperature":26.2,"Humidity":63.9},"TempUnit":"C"}}`

ascillato2, 2019-02-15:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-15 by ascillato2

jacoscar, 2019-02-15:
> sonoffs/cmnd/status

It does actually work from the Tasmota console

```
17:12:49 CMD: sonoffs/cmnd/status 10
17:12:49 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-15T17:12:49","Switch2":"OFF"}}
```

but when the switch is actioned, this is what is published

```
17:13:57 MQT: cmnd/SensorT/POWER2 = ON (retained)
17:13:58 MQT: cmnd/SensorT/POWER2 = OFF (retained)
17:14:00 MQT: cmnd/SensorT/POWER2 = ON (retained)
```

So in my HA configuration.yaml I currently have

```
    state_topic: "cmnd/SensorT/POWER2"
    payload_on: "ON"
    payload_off: "OFF"
```

and to make it work with the response above I would need another state_topic (stat/sonoff/STATUS10) and a value_template, I guess

Jason2866, 2019-02-15:
Turn retain in HA off, and delete the stored retained messages in your broker
read wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-%E2%80%90-Manual-Config

jacoscar, 2019-02-15:
retain should only help when the client goes offline, but in my case is the broker going offline that causes issues...