arendst/Tasmota #5239: MINIMAL firmware - please upgrade

jacoscar, 2019-02-15:
_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ :  Sonoff Basic
- [X ] _Development/Compiler/Upload tools used_ :
- [X ] _If a pre-compiled release or development binary was used, which one?_ :
- [X ] _You have tried latest release or development binaries?_ :
- [X ] _Provide the output of command_``status 0`` :
```
18:35:07 CMD: status 0
18:35:07 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"SensorT","SwitchMode":[2,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":0}}
18:35:07 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"Blocked Loop","Uptime":"0T00:07:07","StartupUTC":"2019-02-15T17:28:00","Sleep":50,"CfgHolder":4617,"BootCount":94,"SaveCount":1836,"SaveAddress":"FB000"}}
18:35:07 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(b4b1ee9-minimal)","BuildDateTime":"2019-02-14T18:15:07","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
18:35:07 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jacopo&Isa",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008089","280500000100000000000000000000000000","00000000"]}}
18:35:07 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":381,"Free":620,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","00082580","000403A2","21021702","000000C0"]}}
18:35:08 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7701","IPAddress":"192.168.1.14","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:6B:1E:15","Webserver":2,"WifiConfig":3}}
18:35:08 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6B1E1515","MqttUser":"Jacopo","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:35:08 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 15 17:35:08 2019","Local":"Fri Feb 15 18:35:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
18:35:08 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-15T18:35:08","Switch2":"ON"}}
18:35:08 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-15T18:35:08","Uptime":"0T00:07:08","Vcc":3.417,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"
```

**To Reproduce**
Trying to update using http://sonoff.maddox.co.uk/tasmota/sonoff.bin or http://sonoff.maddox.co.uk/tasmota/sonoff-minimal.bin or by downloading the file and then selecting it; the main page always shows MINIMAL firmware - please upgrade
I've also tried to update to sonoff-minimal.bin and then to sonoff.bin, but no luck

**Expected behavior**
Device updated and restarted without showing "MINIMAL firmware - please upgrade"

ascillato, 2019-02-15:
Minimal is just an interim version used to free flash memory for the full version. You need to upgrade to the full version.

Use instead the latest version from http://thehackbox.org/tasmota/020500/sonoff.bin

If you want further information about minimal firmware and upgrades, please check the wiki.

If you need further assistant please find us in the Tasmota Support Chat. Thanks

ascillato2, 2019-02-15:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-15 by ascillato2

jacoscar, 2019-02-15:
> http://thehackbox.org/tasmota/020500/sonoff.bin

I've tried that as well but when it restarts, I'm still on the minimal firmware

ascillato, 2019-02-15:
Please, do a reset 5 in the console and then try again.

If that doesn't work, you will need to erase all flash with esptool.py as
explained in the wiki and flash again with the precompiled bins. Sorry.

jacoscar, 2019-02-16:
I don't think reset 5 works; it restarts but it still connects back to the wifi with all the settings as they were

arendst, 2019-02-16:
As long as minimal is loaded you won't be able to change any setting so if your otaurl is set for minimal it won't change to anything else.

Try a firmware upgrade by file upload from the GUI which should work.

After that change your otaurl to the correct binary not being a minimal version.

jacoscar, 2019-02-16:
it still doesn’t work 😫

jacoscar, 2019-02-16:
I ended up reflashing it via serial; I won't try to update it again...

clabnet, 2019-05-23:
Me too. Same issue.

chaoranxie, 2019-08-06:
@clabnet  not sure if you found a solution but this works for me. the answer can be found here https://github.com/arendst/Sonoff-Tasmota/issues/3144  
```
otaurl http://localip:port/data/firmwares/sonoff-full.bin
upload 1
```

couple key point i found is that I have to use the `localip:port` as shown above, if i use an working url like `https://9227d90e.ngrok.io/sonoff-basic.bin` it still does not work. hope this helps a bit

GregWoods, 2019-09-24:
I had a working non-minimal version which was uploaded over serial.
Then did the OTA using the web interface using the file: http://thehackbox.org/tasmota/release/sonoff.bin
It now says I am on minimal.
This is a real issue. (and if the is that I've misunderstood something, then the issue is UI or documentation)

GregWoods, 2019-09-24:
no... the issue is (mostly) me.
Nothing indicated that there was more going on. It actually hadn't finished. Tried again in the time it took to write the above post, and all is well!

meingraham, 2019-09-24:
https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrading#upgrade-by-web-server

In order to OTA upgrade, there has to be enough free program memory to load a new firmware binary along with the existing firmware. The existing firmware is performing the load and when that is done, it hands of to the new firmware. The only way to get two firmware binaries is to make room for them. So, the first thing Tasmota does is load Minimal next to the existing firmware. Once handed off, there is now enough free space to load the ultimate firmware. Minimal loads it, then hands off. Tasmota's web OTA handles this all for you. What you saw was the interim Minimal load which then was immediately followed by the final update.

So, it was... all you ;-)

GregWoods, 2019-09-25:
Yes, all my fault!
However...  the fact that several people have made the same mistake indicates the process could be improved a little.
When the minimal firmware loads, the generic message "minimal firmware, please upgrade" is extremely misleading, and has resulted in wasted time by many.
If the firmware knows it is in the process of flashing the real firmware, the message should indicate this. I know that "every byte counts" with Tasmota, but even removing the "please update" part of the message would be less confusing. Just a couple of tiny ideas to remove as much friction as possible for new users. There is enough to learn as it is :)

meingraham, 2019-09-25:
Because Tasmota is working with limited ESP82xx resources, it re-uses some artifacts to avoid having to store more than is needed in the limited program space. The "Please Upgrade" message is generic. The Minimal software doesn't know how it got there. When a user updates "manually" using the File Upload OTA, they have to manually load Minimal and then repeat the upgrade to load the final firmware. When Minimal is booted, it simply is broadcasting to its world, "I'm not supposed to be here. Do something!" In fact, sometimes a web OTA does barf and the second step doesn't complete successfully. And, in that case, Minimal needs to say the same thing. Thus, knowing that the web OTA is a two step process is left as wiki reading for the user.

Kedryn, 2019-11-28:
I upgraded from 6.5.0.1 2_3_0 and 2_5_0 to 6.5.1. A couple of sonoffs are now on 6.7.1 minimal, but they wond upgrade further. Even if i force them to upgrade, after the reboot they are still on minimal.

Another sonoff keeps staying on 6.5.0.1 2_5_0 and wont even go on 6.7.1 minimal after reboot.

Kedryn, 2019-11-28:
Ok i solved upgrading those with minimal via FILE upload. They are fine now.

lobocobra, 2019-12-06:
minimal firmware will break your device and leave it in an unresponsive state.
=> see also my open issue

lobocobra, 2019-12-06:
Just re-tested it with version 6.5.0.10 and the official minimal-firmware:
http://thehackbox.org/tasmota/tasmota-minimal.bin

What I did....
- on v 6.5.0.10 run "reset 5"
- perform an reboot on the device
- load minimal-firmware
- (as stated by arendst reset 5 does not work now, so I loaded the 7.1.1.
=> nothing happen, I am stuck in the 7.1.1 minimal firmware
=> it shows that like in issue #7120 minimal firmware is broken

lobocobra, 2019-12-06:
A work around is to.....
- compile an own minimal-firmwae load it over the official tasmota-minimal and then load the 7.1.1.

=> something is clearly wrong...

Kedryn, 2019-12-06:
If you are stuck on minimal with OTA, try downloading the normal version, and do an update via file upload. It worked for me

Kedryn, 2019-12-06:
The real problem is the lack of info/log on why an upgrade fails.

stevevg, 2019-12-09:
I have the same issue (stuck on minimal firmware - please upgrade) trying to upgrade a teckin sp22 plug from 6.6.0 to latest 7.1.2
I had already done upgrades before (all in the 6.x major), which went smoothly, but now I'm stuck. Already tried to upload tasmota.bin via file upload, with no luck.

derdickepozilist, 2023-04-14:
Had the same issue with my sonoff basic, switching to tasmota-lite.bin from the official repo did the trick for me.

barbudor, 2024-01-16:
@chopperpl please do not hijack an old issue with a topic barely related
Open a new Discussion instead
And while you are at serial flashing, I suggest directly flashing the latest version, not an old outdated one

chopperpl, 2024-01-16:
@barbudor I have no intention to hijack the topic. The discussion was about issues with OTA updates and claims they are due to the lack of space available on device. I've just chipped in my few words that show issues do exist and based on my experience I have plenty of space left in flash for another minimal firmware.
Thanks for your suggestion. Unfortunately, I have removed all wires from the serial pins before attempting the OTA update and don't feel like spending another x hours to solder them back in. This was very challenging and I don't have precision tools to do so. Therefore, I have gave up on it, but would return, if there is a solution without soldering.
Cheers,

barbudor, 2024-01-16:
Do as you wish but generally strictly following the Doc is a path to success which is is certainly not your case as nowhere we advise to flash v6.6.0
https://tasmota.github.io/docs/Upgrading/#upgrade-flow

We won't answer anymore in that topic