arendst/Tasmota #5244: RFC: Nicer web UI idea offloaded to external server

domenpk, 2019-02-16:
Current web UI as a bit limited, and that's understandable, as there are flash space limitations.

My idea is to source basically offload all the fancier UI to some server.
Now this obviously requires an internet connection, but in case it is not available, you still have the existing UI as a backup. New UI could use SRI hashes, so it couldn't be replaced with malicious UI.
Externally loaded UI offers plenty of possibilities, for example nice timer configuration, rule validations ...
The features would be implemented by sending commands through already implemented web interface.

(I know about TasmoAdmin/SonWEB, which requires local installation, and therefore loses the seamless appeal of this proposal.)

What do you think? Feasible? Does it require only someone implementing it, or are there some restrictions that I have missed?

ascillato, 2019-02-16:
Please, check: https://github.com/jziolkowski/tdm

ascillato, 2019-02-16:
> New UI could use SRI hashes

That would use a lot of RAM resources in the ESP8266 so as to work, leaving nearly no other feature available like sensors support.

domenpk, 2019-02-16:
Was not awre of TDM, but is still something you have to install.

SRI hashes are checked in the web browser.

ascillato, 2019-02-16:
Tasmota is not meant to be connected to internet. The main idea is not to be cloud dependant and that all your information (usage, statistics, data, etc) is just local in your network.

That is why TasmoAdmin Server or TasmoAdmin Add-On for Hass.io are locals in your network.

As an alternative for TasmoAdmin, there is TDM also.

domenpk, 2019-02-16:
Tasmota is still as is, this would not be a dependency. It's just the extra UI web pages that would be transferred over the internet (to the device administering tasmota, which would need internet connectivity for this; in case of no internet, fallback is current behaviour).

ascillato, 2019-02-16:
Any extra tool for Tasmota is always welcome !!!

domenpk, 2019-02-16:
And thanks for that.

I just want(ed) to hear thoughts, and criticism. If someone else thinks this could be useful, I can create a proof of concept, and hopefully people skilled in web design could work on it.

netpok, 2019-02-16:
It's a nice idea, it could be implemented in the same way as the device database I suggested in #5177.

andrethomas, 2019-02-16:
I think most people move away from internet hosted services to get more direct control.

I personally don't like the idea of my devices being dependent on an internet connection and external services that are hosted on a site which is beyond my control.

I do like the idea of being able to configure and control a device over MQTT instead of HTTP.

You asked for thoughts...

ascillato2, 2019-02-17:
Closing this issue as it has been answered.

Please, try a PoC (proof of concept) for this idea. Thanks. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-17 by ascillato2