arendst/Tasmota #5246: [Question]Accurate calibration method

yuuya45, 2019-02-16:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
 [x] Hardware used : Sonoff Pow R2
- [x] If a pre-compiled release or development binary was used, which one? : development binary
- [x] You have tried latest release or development binaries? :Dev 6.4.1.14 20190203

I am struggling to calibrate accurately.
The commands "PowerSet", "VoltageSet" and "CurrentSet" do not seem to accept decimal points.
Is there a way to calibrate correctly?

Is not the command such as "PowerCal" already valid?

arendst, 2019-02-17:
Sonoff Pow R2 does not support PowerCal, VoltageCal and CurrentCal. The were used for more precise calibration but it was felt the following commands could do the same on this device.

It only supports PowerSet, VoltageSet and CurrentSet.

These commands accept decimal points as can be seen here:
```
10:05:41 CMD: voltageset
10:05:41 MQT: stat/pow3/RESULT = {"VoltageSetCal":1969}
10:06:02 CMD: voltageset 224.1
10:06:02 MQT: stat/pow3/RESULT = {"VoltageSetCal":1971}
10:06:15 CMD: voltagecal
10:06:15 MQT: stat/pow3/RESULT = {"Command":"Unknown"}
10:06:35 CMD: voltageset 210.3
10:06:35 MQT: stat/pow3/RESULT = {"VoltageSetCal":1848}
10:06:47 CMD: voltageset 224.6
10:06:47 MQT: stat/pow3/RESULT = {"VoltageSetCal":1971}
10:07:25 CMD: voltageset 210
10:07:25 MQT: stat/pow3/RESULT = {"VoltageSetCal":1862}
10:07:32 CMD: voltageset 210
10:07:32 MQT: stat/pow3/RESULT = {"VoltageSetCal":1871}
10:07:49 CMD: voltageset 222.1
10:07:49 MQT: stat/pow3/RESULT = {"VoltageSetCal":1973}
```

yuuya45, 2019-02-17:
The reason I thought that the decimal point is not valid is from the following result.

    CMD:_ Backlog VoltageSet 100; VoltageSet 101.1; VoltageSet 101.2; VoltageSet 101.3; VoltageSet 101.4; VoltageSet 101.5; VoltageSet 101.6; VoltageSet 101.7; VoltageSet 101.8; VoltageSet 101.9; VoltageSet 102
    22:14:37 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1868}
    22:14:37 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:38 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:38 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:38 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:39 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:39 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:39 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:40 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:40 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1886}
    22:14:40 RSL: stat/sd-p2-br-bed/RESULT = {"VoltageSetCal":1905}
    
    22:16:31 CMD: Backlog PowerSet 60; PowerSet 60.1; PowerSet 60.2; PowerSet 60.3; PowerSet 60.4; PowerSet 60.5; PowerSet 60.6; PowerSet 60.7; PowerSet 60.8; PowerSet 60.9; PowerSet 61; 
    22:16:31 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5383}
    22:16:31 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5383}
    22:16:32 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5386}
    22:16:32 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5386}
    22:16:32 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5380}
    22:16:32 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5376}
    22:16:33 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5376}
    22:16:33 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5390}
    22:16:33 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5390}
    22:16:34 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5387}
    22:16:34 RSL: stat/sd-p2-br-bed/RESULT = {"PowerSetCal":5477}
    
    22:25:49 CMD:  Backlog CurrentSet 5000; CurrentSet 5000.1; CurrentSet 5000.2; CurrentSet 5000.3; CurrentSet 5000.4; CurrentSet 5000.5; CurrentSet 5000.6; CurrentSet 5000.7; CurrentSet 5000.8; CurrentSet 5000.9; CurrentSet 5001; 
    22:25:49 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144060}
    22:25:50 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144045}
    22:25:50 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144045}
    22:25:50 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144045}
    22:25:50 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144045}
    22:25:51 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144045}
    22:25:51 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144045}
    22:25:51 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144045}
    22:25:52 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144045}
    22:25:52 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144100}
    22:25:52 RSL: stat/sd-p2-br-bed/RESULT = {"CurrentSetCal":144128}

Voltage is not changing from 0.1 to 0.9, and power is always set up and down.
(Current is not very problematic because it only adjusts power factor = 1, but it is not accurate.)

There is also a slight voltage fluctuation so VoltageSet always goes up and down the offset.

So whether you can set it finely with decimal point,
I hope I can adjust the offset directly with VoltageCal.

By the way, this problem is important because I have to proofread many Pow and S31 (over 100) from now on.

arendst, 2019-02-17:
You may be right!!!

I looked at the code and I think I made an error here:
```
      Settings.energy_voltage_calibration = ((unsigned long)CharToDouble(XdrvMailbox.data) * voltage_cycle) / CSE_UREF;
```
it should be like
```
      Settings.energy_voltage_calibration = (unsigned long)(CharToDouble(XdrvMailbox.data) * voltage_cycle) / CSE_UREF;
```
I'll change in dev branche.

yuuya45, 2019-02-17:
Thank you for your quick response!

I confirmed that the bug was corrected now and it was able to be set accurately.

closed 2019-02-17 by yuuya45