arendst/Tasmota #5252: Flickering and flag for disabling sleep in light scheme 0

netpok, 2019-02-17:
With commit f23d18723e1cbd469ac95a51d3c1aa910c3377e6 the sleep became enabled when using light scheme 0. While in most cases this is preferred to save power, it is causing flickering with devices using substandard power supply like SP10 smart plug.

The root of the problem is that the change in power consumption of ESP directly affects the rail voltage, so while the pwm output is stable the driving current varies. This can be solved with executing sleep 0 on the device, but as result the device will not sleep even with lights off.

I'm proposing the introduction of device flag to disable light scheme 0 sleep, because I think this can affect many devices (mainly smart plugs with included light features).

Also It needs checking, but is PWM working with the old sleep? If not a sleep method check should be also included in condition.

andrethomas, 2019-02-17:
It can be done but should be limited to cases where chip based pwm RGB like the WS2812 and SM16716 is not used.

The problem which was mitigated with the original PR (https://github.com/arendst/Sonoff-Tasmota/commit/f23d18723e1cbd469ac95a51d3c1aa910c3377e6) and eventually implemented with merge (https://github.com/arendst/Sonoff-Tasmota/commit/f8350d65c4832077020b926556a8c5b651ccbfb3) is with sleep 0 unpredictable behaviour is introduced on devices such as the Sonoff Basic R1 and Wemos D1 mini boards that have an inefficient power supply and/or heat dissipation design.

The other solution is to use 2.5.0 core based binaries as I think those should be using hardware pwm.

arendst, 2019-02-17:
PWM support before 2.5.0 and since 2.5.0 is the same; previous tasmota core versions used the code form core 2.4.0 which had NMI support added which in turn solved issues in core 2.3.0, 2.4.1 and 2.4.2.

Why not revert the change? Or keep sleep 0 for PWM but enable sleep for scheme 0 on WS2812?

andrethomas, 2019-02-17:
> Why not revert the change? Or keep sleep 0 for PWM but enable sleep for scheme 0 on WS2812?

WS2812 support is enabled by default so the latter would only work in cases where WS2812 would be disabled during compile time?

netpok, 2019-02-17:
This problem is not pwm related. The leds aren't flickering because the PWM signal is bad, when the chip goes to sleep it's current draw is significantly decreased, so the rail voltage increases leading the leds becoming brighter. It is possibly affecting WS2812 based devices too.

With devices using proper power supply, this is not a problem, because the leds using way more power then the controlling device itself. The problematic devices are the smart plugs, which are using non isolated step down supplies, the led brightness are also affected by the state of the relay, but that can't be helped and it is not a millisecond pattern.

andrethomas, 2019-02-17:
> when the chip goes to sleep it's current draw is significantly decreased, so the rail voltage increases leading the leds becoming brighter.

The difference between dynamic sleep 0 and sleep 50 is only about 50mA at best so if this is causing rail voltage changes then the problem is not necessarily with sleep.

netpok, 2019-02-17:
Sadly it is enough because the relay also affects the brightness and that is not using more then 60-100 mA.

andrethomas, 2019-02-17:
> Sadly it is enough because the relay also affects the brightness and that is not using more then 60-100 mA.

How are you powering the led strip?

netpok, 2019-02-17:
This is not a led strip this is an SP-10 smart plug with nightlight function included, there are 8-12 leds driven directly from the 5V rail powered by a stepdown converter.

andrethomas, 2019-02-17:
Interesting, so it appears they did not design the 5V rail according to the current requirements of the 8-12 led's and the relay because if they did you should not see a brightness difference between sleep 0 and sleep 50 and/or relay switching if there is enough current to sufficient power the 8-12 LEDs and the relay with stability.

It's easy to revert the change but its concerning what impact it will have on the life expectancy of the device since the current sink when the relay is switched on causes a voltage drop on the 5V rail.

netpok, 2019-02-17:
That's why I proposed a device flag instead of a revert.

Also I think it's not that they did not design for the current of the leds, it's probably more like they didn't design for the missing current draw as some buck converters have minimal output current requirement for stable output voltage.

andrethomas, 2019-02-18:
@netpok 

I've proposed PR https://github.com/arendst/Sonoff-Tasmota/pull/5264 which effectively reverts the original change that left sleep enabled for scheme 0 but it still gives the option to compile for how it is now by un-commenting '#PWM_LIGHTSCHEME0_IGNORE_SLEEP` in sonoff.h

By default it is commented out so sleep is disabled for scheme 0

netpok, 2019-02-18:
Since this issue is directly related to a group of specific device I still think it would be better off with a device flag. Most affected user would just simply use `sleep 0` instead of recompiling the firmware, probably me too, because OTA upgrades are nice.

andrethomas, 2019-02-18:
It will automatically enter sleep 0 as it did before the merge you cited.

netpok, 2019-02-18:
I misread the condition, but still thinks this should be runtime configurable. The only question in my opinion is: does this worth the use of 1 of the 8 device flags? If not than your PR is the way.

arendst, 2019-02-18:
@netpok thx for keeping an eye on the flags too ;-). For now I choose the PR.

netpok, 2019-02-18:
In that case I'm closing this for now.

closed 2019-02-18 by netpok