arendst/Tasmota #5255: Drop NO_PULLUP flag

netpok, 2019-02-17:
I think before the release of the template system, the NO_PULLUP flag should be dropped. It is a flawed solution for the Shelly2, because at the time of implementation it could have been solved with an already existing solution.

The current Shelly2 configuration is:
```  { "Shelly 2",        // Shelly2 (ESP8266 - 2MB) - https://shelly.cloud/shelly2/
     0,
     GPIO_MCP39F5_TX,  // GPIO01 MCP39F501 Serial input
     0,
     GPIO_MCP39F5_RX,  // GPIO03 MCP39F501 Serial output
     GPIO_REL1,        // GPIO04
     GPIO_REL2,        // GPIO05
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_SWT1,        // GPIO12
     0,
     GPIO_SWT2,        // GPIO14
     GPIO_MCP39F5_RST, // GPIO15 MCP39F501 Reset
     0,
     GPIO_FLAG_PULLUP  // Allow input pull-up control
  }```

but the same can be achieved with replacing the switch definition with the proper no pullup version:
```  { "Shelly 2",        // Shelly2 (ESP8266 - 2MB) - https://shelly.cloud/shelly2/
     0,
     GPIO_MCP39F5_TX,  // GPIO01 MCP39F501 Serial input
     0,
     GPIO_MCP39F5_RX,  // GPIO03 MCP39F501 Serial output
     GPIO_REL1,        // GPIO04
     GPIO_REL2,        // GPIO05
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_SWT1_NP,        // GPIO12
     0,
     GPIO_SWT2_NP,        // GPIO14
     GPIO_MCP39F5_RST, // GPIO15 MCP39F501 Reset
     0,
     0  // Allow input pull-up control is not needed since _NP switches are used
  }```

After releasing the template support, it cannot be removed easily, because of backward compatibility reasons.

arendst, 2019-02-17:
I agree. Will get rid of it now.

closed 2019-02-17 by netpok

localhost61, 2019-02-18:
@arendst 
Then D_ALLOW_PULLUP is not needed in language files anymore.

arendst, 2019-02-18:
@localhost61 just leave them be. One never knows I come up with another use case.

localhost61, 2019-02-18:
OK, but just to mention, translation is not always literal, it happens to be contextual for reducing size, especially here. It may not be adequate in an other context.

arendst, 2019-02-18:
@localhost61 I agree. But as long as the text string is not used in the code it does not take up any code space so no problem if it is still in the language file.

localhost61, 2019-02-18:
@arendst 
Size is not only for code is for GUI too. In French, or German, it's not easy to be concise and stay meaningful ;-) 
For instance if I literally translate `Flags` into `Drapeaux` I'm not sure french people will understand what it's all about, so I prefer `Options`.

arendst, 2019-02-18:
Agree. I will change the not-translated Flags yet to Options now.